mne-tools/mne-python #7134: Cannot import write_raw_bids from mne_bids

pchholak, 2019-12-11:
#### Describe the bug
When importing 'write_raw_bids' from 'mne_bids', Attribute Error is produced suggesting mne.io has no attribute 'read_raw_bdf'.

#### Steps to reproduce
```
from mne_bids import write_raw_bids
```

#### Actual results
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-8-cb838460d511> in <module>
      6 from mne.datasets import sample
      7 
----> 8 from mne_bids import write_raw_bids
      9 # from mne_bids import read_raw_bids
     10 # from mne_bids import make_bids_basename

~\Anaconda3\lib\site-packages\mne_bids\__init__.py in <module>
      4 
      5 
----> 6 from mne_bids.write import (write_raw_bids, make_bids_folders, make_bids_basename,  # noqa: E501 F401
      7                             make_dataset_description, write_anat)  # noqa: F401
      8 from mne_bids.read import read_raw_bids, get_head_mri_trans  # noqa: F401

~\Anaconda3\lib\site-packages\mne_bids\write.py in <module>
     32                             _get_mrk_meas_date, _extract_landmarks, _parse_ext,
     33                             _get_ch_type_mapping)
---> 34 from mne_bids.copyfiles import (copyfile_brainvision, copyfile_eeglab,
     35                                 copyfile_ctf, copyfile_bti)
     36 from mne_bids.read import reader

~\Anaconda3\lib\site-packages\mne_bids\copyfiles.py in <module>
     22 from scipy.io import loadmat, savemat
     23 
---> 24 from mne_bids.read import _parse_ext
     25 
     26 

~\Anaconda3\lib\site-packages\mne_bids\read.py in <module>
     27           '.fif': io.read_raw_fif, '.pdf': io.read_raw_bti,
     28           '.ds': io.read_raw_ctf, '.vhdr': io.read_raw_brainvision,
---> 29           '.edf': io.read_raw_edf, '.bdf': io.read_raw_bdf,
     30           '.set': io.read_raw_eeglab}
     31 

AttributeError: module 'mne.io' has no attribute 'read_raw_bdf'

agramfort, 2019-12-11:
you must have an install problem.

how did you install mne and mne_bids? what versions are you using?

>

pchholak, 2019-12-11:
I installed using
```
conda install -c conda-forge mne-bids
```

MNE version is 0.19.2

agramfort, 2019-12-11:
can you tell me what mne.sys_info() gives you?

>

pchholak, 2019-12-11:
Platform:      Windows-10-10.0.18362-SP0
Python:        3.7.1 (default, Dec 10 2018, 22:54:23) [MSC v.1915 64 bit (AMD64)]
Executable:    C:\Users\Usuario\Anaconda3\python.exe
CPU:           Intel64 Family 6 Model 158 Stepping 10, GenuineIntel: 12 cores
Memory:        31.9 GB

mne:           0.19.2
numpy:         1.16.1 {blas=C:\\projects\\numpy-wheels\\numpy\\build\\openblas, lapack=C:\\projects\\numpy-wheels\\numpy\\build\\openblas}
scipy:         1.1.0
matplotlib:    3.0.2 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.20.1
numba:         0.41.0
nibabel:       2.5.1
cupy:          Not found
pandas:        0.23.4
dipy:          1.0.0
mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.9.2}
pyvista:       Not found
vtk:           8.2.0

agramfort, 2019-12-11:
can you try

from mne.io import read_raw_bdf

>
in a fresh terminal

pchholak, 2019-12-11:
I tried importing read_raw_bdf from mne.io and it works fine.

pchholak, 2019-12-11:
UPDATE: The entire issue is solved after restarting jupyter notebook.

closed 2019-12-11 by agramfort