mne-tools/mne-python #7136: VIZ: bug in plot_sensors(kind='3d')

drammock, 2019-12-11:
#### Describe the bug
When plotting MEG and EEG sensors at the same time, the EEG and MEG sensors appear in the wrong position relative to one another.  Presumably this is an issue of which coordinate frame is chosen.  An example is [here](https://mne.tools/dev/auto_tutorials/epochs/plot_20_visualize_epochs.html#plotting-sensor-locations), where the EEG sensors are higher up than the MEG sensors (implying that the subject's head was embedded in the MEG helmet).

#### Steps to reproduce
See [the linked tutorial](https://mne.tools/dev/auto_tutorials/epochs/plot_20_visualize_epochs.html#plotting-sensor-locations).

@agramfort WDYT?  @larsoner suggested (offline) that if `ch_type='all'` then everything should get transformed to head coords, but if only plotting a single sensor type we can plot in whatever coord sys they're in.  If the overhead is not too bad, I might suggest always plotting in head coords, even if only one sensor type is chosen.  As an aside, it might also be good to label the 3d plot axes with a unit (meters).

agramfort, 2019-12-11:
+1 with what @larsoner suggested.

>

larsoner, 2019-12-12:
> If the overhead is not too bad

Transform overhead should be negligible.

> I might suggest always plotting in head coords, even if only one sensor type is chosen.

I was actually starting to think the same thing. So I would say to always use head coords for now, and if people want MEG sensors in MEG coords we can add `coord_frame` later -- or just tell them to use `plot_alignment` which already has these transform options built in.

>  As an aside, it might also be good to label the 3d plot axes with a unit (meters).

Agreed, feel free to add this at the same time as the coord frame changes

PR #7142 (2019-12-15): MRG, FIX: apply transform before plotting sensors in 3D
closes #7136 

Not sure if this is robust enough to handle all use cases...  @larsoner are sensor coords ever going to be in some other space besides `dev` or `head`?  If so maybe I need a trans dict that maps from the `info['chs'][n]['coord_frame']` integer to the appropriate transform in each case.
closed 2019-12-15 by drammock