mne-tools/mne-python #7140: CSD multitaper: should we keep single tapers ?

brovelli, 2019-12-13:
Dear all,

I have a question regarding the computation of the CSD using multitapers.

The [def _csd_from_mt](https://github.com/mne-tools/mne-python/blob/20195a24481241ef38d854ed86dea0f68dcffd5b/mne/time_frequency/multitaper.py#L253) seems to return the average CSD across tapers.

However, as far as I understand the methods (see [Mitra and Pesaran, 2009](https://www.ncbi.nlm.nih.gov/pmc/articles/PMC1300074/pdf/9929474.pdf)) the PSD is computed by averaging the power estimates computed for each taper (eq. 6 and 7).

Questions:

1) Do you think that the two approaches are equivalent ? 

2) If yes, forget this issue.

3) If not, I would suggest to keep the single tapers complex CSD. However, the PSD should be computed for each taper and then averaged.

I am asking this, because the correct implementation of the PSD using mtm could be relevant also for source estimated using DICS and mtm.

Thanks a lot

larsoner, 2019-12-13:
@wmvanvliet can you comment?

wmvanvliet, 2019-12-13:
Sorry, I know nothing about multi-tapers. I always use wavelets to compute de CSD.

agramfort, 2019-12-17:
the diag of the CSD matrix is the PSD. So averaging the CSDs is compatible with averaging of the PSDs as done with MT. Makes sense?

brovelli, 2019-12-17:
Well, however, the CSD is complex-valued, whereas the PSD is not. So averaging the CSD across tapers and then take the power should be different from taking the PSD and then avereage the power.....

In practice, a colleague of mine has now tried the two approaches for the calculcation of source-power using DICS with multitapers and she finds practically identical (at the 10^-20 resolution).

So, this may not be an issue at all. In fact, I think I remember Fieldtrip does the same thing as MNE

agramfort, 2019-12-17:
the diag is real valued:

In [13]: np.diag(csd_mt.get_data(index=0)).imag

Out[13]:

array([0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,

       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,

       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,

       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,

       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,

       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,

       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,

       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,

       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,

       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,

       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,

       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.])

>

brovelli, 2019-12-17:
Ahhh of course ! Super Alex ! Thanks

Could it still be an issue for the calculation of spectral coherence based on multitapers using mne.connectivity.spectral_connectivity since the off-diagonal terms are used ?

I guess that Denis and Martin have been checking this already...

agramfort, 2019-12-17:
I think we are good as it is.

>

brovelli, 2019-12-17:
Great! I think we can close this issue.

agramfort, 2019-12-17:
please close

closed 2019-12-17 by brovelli