mne-tools/mne-python #7149: How to export data in .set file

sabu3, 2019-12-18:
I import .set file through mne.io.read_raw_eeglab function to modify some value in the imported data.
Then I want to export modified data in .set format to analyze data on EEGLAB.
But I cannot find the function to export data in .set format.
Please give me advises and comments.
It would be very helpful if I can get any suggestions.

cbrnr, 2019-12-18:
Currently MNE doesn't have an option to export to .set. However, you could use [MNELAB](https://github.com/cbrnr/mnelab) (or copy the relevant code).

sabu3, 2019-12-18:
Thank you for your immediate reply.
I will try MNELAB to export to .set.

closed 2019-12-28 by agramfort

jackz314, 2021-03-11:
For those looking to integrate the MNELAB solution with MNE, I believe the relevant code is [here](https://github.com/cbrnr/mnelab/blob/d20722d81728953182682a99f116cecbaaa15945/mnelab/io/writers.py#L19)