mne-tools/mne-python #7154: Issue with pick_types(emg=True)

TariqCannonier, 2019-12-19:
#### Describe the bug
We get an error of "No Data channels found"  when selecting emg channels with `picks=['emg']` in a dataset which contains only emg channels. 

#### Steps to reproduce

together with @jasmainak and @simonatemereanca we wrote this script:
https://gist.github.com/TariqCannonier/0c708ae01268bfbb766bd320eac45035
to reproduce the bug.

#### Expected results
Should plot PSD for EMG channels only

#### Actual results
```py

  File "<ipython-input-7-ad302adea22f>", line 1, in <module>
    runfile('/Users/tariqcannonier/.spyder-py3/temp.py', wdir='/Users/tariqcannonier/.spyder-py3')

  File "/Users/tariqcannonier/anaconda3/lib/python3.7/site-packages/spyder_kernels/customize/spydercustomize.py", line 827, in runfile
    execfile(filename, namespace)

  File "/Users/tariqcannonier/anaconda3/lib/python3.7/site-packages/spyder_kernels/customize/spydercustomize.py", line 110, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "/Users/tariqcannonier/.spyder-py3/temp.py", line 20, in <module>
    raw.plot_psd(picks=['emg'])

  File "</Users/tariqcannonier/anaconda3/lib/python3.7/site-packages/mne/externals/decorator.py:decorator-gen-147>", line 2, in plot_psd

  File "/Users/tariqcannonier/anaconda3/lib/python3.7/site-packages/mne/utils/_logging.py", line 90, in wrapper
    return function(*args, **kwargs)

  File "/Users/tariqcannonier/anaconda3/lib/python3.7/site-packages/mne/io/base.py", line 1585, in plot_psd
    spatial_colors=spatial_colors, verbose=verbose)

  File "</Users/tariqcannonier/anaconda3/lib/python3.7/site-packages/mne/externals/decorator.py:decorator-gen-133>", line 2, in plot_raw_psd

  File "/Users/tariqcannonier/anaconda3/lib/python3.7/site-packages/mne/utils/_logging.py", line 90, in wrapper
    return function(*args, **kwargs)

  File "/Users/tariqcannonier/anaconda3/lib/python3.7/site-packages/mne/viz/raw.py", line 621, in plot_raw_psd
    raw.info, proj, picks, ax, area_mode)

  File "/Users/tariqcannonier/anaconda3/lib/python3.7/site-packages/mne/viz/utils.py", line 3005, in _set_psd_plot_params
    raise RuntimeError('No data channels found')

RuntimeError: No data channels found
```

#### Additional information
〈Replace this text with the output of `mne.sys_info()`.〉

agramfort, 2019-12-20:
pb is in _set_psd_plot_params

we could change the logic there. After doing picks = _picks_to_idx(info,
picks)
we get all the types present in picks and work with these rather and
enforcing
some kind of data channel

wdyt?

larsoner, 2019-12-20:
Sounds like the right thing to do

TariqCannonier, 2019-12-20:
I also want to add that a similar issue exists when attempting to filter that EMG channel, although it does not result in an error.
```py
raw.filter(5, 25, picks=['emg'])
```
Will output:
"No data channels found. The highpass and lowpass values in the measurement info will not be updated."

Seems like setting types EMG is not recognized as a data channel .

agramfort, 2019-12-20:
2 ways out here:

- make emg a data channel which therefore will appear in all plots
when we set picks=None
- fix the logic if these functions as I agree it should work

larsoner, 2019-12-20:
> a similar issue exists when attempting to filter that EMG channel, although it does not result in an error.

Does it actually filter the channel? It should. The message is meant to be only about whether or not the `info` fields get updated, not whether or not your `picks` actually get filtered.

TariqCannonier, 2019-12-20:
It did filter. I thought the message was an error that the channel could not be filtered, even though my data was.  I was a quite confused.

jasmainak, 2019-12-20:
I want to add here that the notion of "data channels" is not clarified in the docs. I think we should just say "EEG and MEG channels". Probably could be fixed as part of the same PR.

larsoner, 2019-12-20:
The problem is that it's not just M/EEG, but also (at least) sEEG, ECoG, and fNIRS. So I'd rather say in the docs (maybe glossary?) what "data channels" are, that way if we add a new "data channel" we have fewer places that need updating.

larsoner, 2020-04-27:
@drammock fixed this in #7668

closed 2020-04-27 by larsoner