mne-tools/mne-python #7159: Compute all-to-all connectivity in sensor space issue

Charlesi8, 2019-12-26:
hi, 
 i just ran the Compute all-to-all connectivity in sensor space .py but here came a ValueError :'zero-size array to reduction operation maximum which has no identity'  . I cannot fix it ,so could someone help me please.
[image]

agramfort, 2019-12-26:
is it running just the example script online or using your own data?

Charlesi8, 2019-12-26:
i downloaded the example script without amendment then ran.

agramfort, 2019-12-26:
I don't have the problem. Can you say what

mne.sys_info()

gives you?

Charlesi8, 2019-12-27:
[image]

agramfort, 2019-12-27:
sorry I cannot replicate. Can you ask a friend to replicate with his
machine? or maybe can
you try on a different machine?

>

Charlesi8, 2019-12-27:
ok, thanks for your advice , and i am thinking to run on another machine.

larsoner, 2020-03-04:
Closing due to non-response. FWIW the reduction error happens when the array is empty, i.e., `con_val` must be a zero-length array.

closed 2020-03-04 by larsoner