mne-tools/mne-python #7161: AttributeError: 'Figure' object has no attribute 'lasso'

lxdfrank, 2019-12-27:
**READ THIS FIRST:** When I run the plot_40_sensor_locations.py, a question of no attribute 'lasso' is appearing. and how to solve it

agramfort, 2019-12-27:
is it an error or a warning?

matplotlib backend are you using? can you report what mne.sys_info()
outputs?

>

lxdfrank, 2019-12-27:
[image]
This is the error result。 The Figure object has no attribute lasso

lxdfrank, 2019-12-27:
> is it an error or a warning? matplotlib backend are you using? can you report what mne.sys_info() outputs?
> […](#)

[image]
This is the error result。 The Figure object has no attribute lasso

agramfort, 2019-12-27:
can you make sure you use qt matplotlib backend and that you have a recent
matplotlib version?

>

lxdfrank, 2019-12-27:
the version of matplotlib is 3.1.1

agramfort, 2019-12-27:
what does mne.sys_info() give you?

I suspect it comes from a weird choice of matplotlib backend

lxdfrank, 2019-12-27:
> what does mne.sys_info() give you? I suspect it comes from a weird choice of matplotlib backend

[image]
and how to modify

agramfort, 2019-12-27:
can you start ipython and from there run

%run plot_40_sensor_locations.py

this will run the example in interactive mode.

>

larsoner, 2020-03-04:
Closing due to non-response

closed 2020-03-04 by larsoner