mne-tools/mne-python #7162: New interface for _TimeViewer

GuillaumeFavelier, 2019-12-28:
This PR summarizes the ideas to implement in `_TimeViewer` (suggested in https://github.com/mne-tools/mne-python/pull/7153#issuecomment-569378110):

**ToDo**

- [ ] Add vertical orientation to the scalarbar (suggested in https://github.com/mne-tools/mne-python/pull/7153#issuecomment-568758879) 
- [ ] Use button widgets (suggested in https://github.com/mne-tools/mne-python/pull/7153#issuecomment-569346780)
- [ ] Refresh the UI design (suggested in https://github.com/mne-tools/mne-python/pull/7153#issuecomment-569315991)
- [ ] Use PySurfer-like time dilation (suggested in https://github.com/mne-tools/mne-python/pull/7153#issuecomment-569346365)
- [ ] Bring colorbar values auto-scaling (suggested in https://github.com/mne-tools/mne-python/pull/7153#issuecomment-569261106)
- [ ] Add support for `hemi=split` (suggested in https://github.com/mne-tools/mne-python/pull/7153#issuecomment-569346365)

**Bug**

- [ ] `dorsal` view causes the `_Brain` object to disappear (reported in https://github.com/mne-tools/mne-python/pull/7153#issuecomment-568735444)

hoechenberger, 2020-02-13:
@GuillaumeFavelier 
regarding this bug report:
> * [ ]  The help window opens in the background in `jupyter qtconsole` (reported in [#7305 (comment)](https://github.com/mne-tools/mne-python/pull/7305#issuecomment-585748906))

for me it's the _PyVista_ window that opens in the background. once activated, pressing ? opens the help window in the foreground.

GuillaumeFavelier, 2020-02-14:
The recommended version of `pyvista` to use is now `0.23.1`. The `master` branch is not supported at the moment so it will break with an error related to `interactor`. This change is made to improve stability.

larsoner, 2020-02-14:
So we hold off on merging this until PyVista cuts a new version?

In that case you might as well add some more stuff here, such as the time vertical bar plus clicking. That would at least bring this to feature parity with `plot_volume_source_estimates`, which is a reasonable stopping point.

larsoner, 2020-02-14:
Argh sorry I thought this was #7247. My comment pertains to that PR :)

GuillaumeFavelier, 2020-02-14:
> So we hold off on merging this until PyVista cuts a new version?

I think it can be merged after some tests are added. The important blockers are https://github.com/mne-tools/mne-python/issues/7228

SophieHerbst, 2020-03-04:
Another small thing I realized: stc.plot allows to set a title (`title = my_title`), which I don't see anywhere on the figure.

GuillaumeFavelier, 2020-03-04:
Oh my! You're totally right, I'm on it. This is easy to fix.

larsoner, 2020-03-11:
@GuillaumeFavelier I tried changing `plot_parcellation.py` to use label-by-label adding but it was terribly slow (like half a second per label and there are 400+ labels in one of the three parcellations that gets plottted):

https://gist.github.com/larsoner/67c09029f47cd496a8414b0a98de857f

So it's probably better to actually implement `plot_parcellation` by doing whatever PySurfer does.

GuillaumeFavelier, 2020-03-11:
These are valuable infos. Thanks for looking into this, I'll follow your advice.

GuillaumeFavelier, 2020-03-19:
The recommended version of pyvista to use is now `0.24`. The previous version `0.23.1` is not supported anymore because of an API change related to interactor. This change is made to improve stability.

hoechenberger, 2020-03-26:
As I ditched my mouse long long ago and prefer trackpads / touchpads these days, I just discovered that I really miss and would so much love to have: pinch to zoom.

GuillaumeFavelier, 2020-03-26:
Qt supports it although I don't now if it's consistent across all operating systems. I updated the list with this feature request for **0.21** :+1:

larsoner, 2020-03-26:
+1 for adding this preferably at the PyVista level. It seems to exist in VTK but might only be in the nightly (having trouble finding the 8.x stable docs):

https://vtk.org/doc/nightly/html/classvtkInteractorStyle.html#a69c0488e202fa20cd82c99d5c76aaea0

GuillaumeFavelier, 2020-03-26:
This could be a good idea for `pyvista/pyvista-gui` actually...

hoechenberger, 2020-03-29:
@GuillaumeFavelier I'm horrible with brain anatomy. I would be super awesome if I could simply click on or hover over a vertex to retrieve the associated `Label` of the respective ROI. This would also make MNE a great tool for teaching brain anatomy!!

hoechenberger, 2020-03-29:
@GuillaumeFavelier Another one.

I typically use the GFP to decide at which time periods to look at source activity. Now with the new traces viewer, it should theoretically be possible to display the GFP trace, or sensor signal of single electrodes. Since the STC doesn't contain the sensor signals anymore (to my understanding), one could modify the STC classes to retain that data. But it seems cumbersome and error-prone, as it would raise the question which sensor data the visualization e.g. of only one hemisphere should retain: data for both hemispheres, or just of the displayed hemisphere? etc

So maybe the easiest would simply be, allow the user to pass a `Raw` or `Evoked` object, crop it to the same time as the STC, and plot the requested traces. Or even expect the users to crop manually beforehand, and supply an object of the correct size.

```python

evoked = ...
stc = mne.minimum_norm.apply_inverse(evoked, ...)

gfp = (evoked.copy()
       .pick_types(eeg=True, meg=False)
       .data
       .std(axis=0))

FPz = (evoked
       .copy()
       .picks('FPz')
       .data)

brain = stc.plot(...)
brain.add_traces([dict(label='GFP', data=gfp),
                  dict(label='FPz', data=FPz)])
```

agramfort, 2020-03-29:
I can show you how to do this with freeview

hoechenberger, 2020-03-29:
@agramfort Nice, let's do this :)
But it would still be great if I could do this in MNE directly :) 

@GuillaumeFavelier Can I continue going crazy a little? I'd love to be able to select a time range in the traces pane, and the displayed activity would be the averaged activation over this time period. Just like in `mne.viz.plot_evoked`.

larsoner, 2020-05-28:
Added #7775 to the list

larsoner, 2020-09-11:
@GuillaumeFavelier it looks like the only remaining items for 0.21 are:

- jupyter notebook integration
- Reduce pyvista viz test time

I'm not sure the notebook integration with multi-views is critical here. @agramfort you okay to push this to 0.22? I'm a bit worried about trying to add this feature right before release. Seems like it would be better to have some time to test it.

agramfort, 2020-09-11:
yes for 0.22

>

larsoner, 2021-04-05:
Maybe after this it might not be too difficult to add support for shadows:

https://github.com/pyvista/pyvista/pull/1247

Feel free to add to the list above or give it a shot if you agree @GuillaumeFavelier . At some point we will want to check that shadows + EDL + SSAO all get along, too.

akaszynski, 2021-04-05:
> Feel free to add to the list above or give it a shot if you agree @GuillaumeFavelier . At some point we will want to check that shadows + EDL + SSAO all get along, too.

I'm going to work on getting the PR working, but after that I'll work on getting multi-pass working according to [MultiPass Rendering](https://vtk.org/Wiki/VTK/MultiPass_Rendering).

larsoner, 2022-02-11:
@GuillaumeFavelier can you bump all the 1.0 stuff to 1.1 after making sure none of it is critical?

GuillaumeFavelier, 2022-02-11:
Sure, will do

larsoner, 2023-09-25:
I think we can close this and follow with targeted issues/PRs

closed 2023-09-25 by larsoner