mne-tools/mne-python #7192: mne.io.read_epochs_eeglab inconsistent behaviour

Davi1990, 2020-01-10:
Dear MNE expert, 

I have run mne.io.read_epochs_eeglab and everything went smoothly with the majority of my data sample. However, for some of them I got the following error: 

[Errno 2] No such file or directory: path/to/file.fdt

Apparently for some reason it doesn't see the file that it is of course in the same path. 

Do you have any tips?

Thanks 

Dave

drammock, 2020-01-10:
If you've copied the error exactly, then it means that Python is looking for a file called `file.fdt` in a location `./path/to/` (relative to the current working directory where Python was started).  That folder location is almost certainly not correct, and probably reflects some example code that you copy-pasted and forgot to comment out or remove from your script.

If that error message was just an example: well, Python is usually not wrong about whether or not a file exists.  Try commands like `os.getcwd()` and `os.listdir()` to make sure the Python interpreter is in fact working in the directory that you think it's in, and whether the file is in fact where Python is looking.

Davi1990, 2020-01-10:
Dear Daniel, 

thanks for your reply. 

I double checked thousands times and the file is in the corresponding folder. All other files are in the same folder and the problem as I said is for only few subjects. 

Anyway I checked it out another time...I have run the `os.listdir()` and the file is in there. 

Any other suggestions? Shall I share both files? 

Thanks

drammock, 2020-01-10:
if the error is "no such file or directory" then sharing the file itself won't help.  Can you edit lines 3 and 4 below, then run this and tell us the output?

```
import os
import mne
folder_where_files_are = os.path.join('path', 'to', 'files')  # <- edit this
full_file_path = os.path.join(folder_where_files_are, 'file.fdt')  # <- edit this
print(os.getcwd())
print(os.listdir(folder_where_files_are))
mne.io.read_epochs_eeglab(full_file_path)
```

Davi1990, 2020-01-10:
I've run the command. 
This is the error I got

TypeError: Expecting miMATRIX type here, got 4002551232

I haven't shared the ouput of `print(os.listdir(folder_where_files_are))` because I have a lot of files in that folder and It would be unreadable. However I can assure you that the file is in there. 

Thanks

drammock, 2020-01-10:
that is a different error from "no such file or directory". Anyway if you don't paste the full output (traceback, etc) there's little I can do to help.

Davi1990, 2020-01-10:
`import os 
import mne
folder_where_files_are = os.path.join('/Volumes/WD_Elements/BROAD/EEG/M1_IPL/Recep/M1_IPL/Visit_1/M1')
full_file_path = os.path.join(folder_where_files_are, '80LP_V1_SP_M1_S10_Final.fdt') 
print(os.getcwd())`
                                                                                                                                                                                                                                    
/Volumes/WD_Elements/BROAD/EEG/M1_IPL/Recep/M1_IPL/Visit_1/M1

`print(os.listdir(folder_where_files_are))` 

['78TW_V1_150SP_M1_S10_Final.fdt', '._78TW_V1_150SP_M1_S10_Final.fdt', '78TW_V1_150SP_M1_S10_Final.set', '._78TW_V1_150SP_M1_S10_Final.set', '80LP_V1_SP_M1_S10_Final.fdt', '._80LP_V1_SP_M1_S10_Final.fdt', '80LP_V1_SP_M1_S10_Final.set', '._80LP_V1_SP_M1_S10_Final.set', '81SSF_V1_SP_M1_S10_Final.fdt', '._81SSF_V1_SP_M1_S10_Final.fdt', '81SSF_V1_SP_M1_S10_Final.set', '._81SSF_V1_SP_M1_S10_Final.set', '82YZ_V5_SP_M1_S10_Final.fdt', '._82YZ_V5_SP_M1_S10_Final.fdt', '82YZ_V5_SP_M1_S10_Final.set', '._82YZ_V5_SP_M1_S10_Final.set', '83KW_V1_SP_M1_S10_Final.fdt', '._83KW_V1_SP_M1_S10_Final.fdt', '83KW_V1_SP_M1_S10_Final.set', '._83KW_V1_SP_M1_S10_Final.set', '84LAL_V2_150sp_M1_S10_Final.fdt', '._84LAL_V2_150sp_M1_S10_Final.fdt', '84LAL_V2_150sp_M1_S10_Final.set', '._84LAL_V2_150sp_M1_S10_Final.set', '52SC_V1_120SP_M1_Merged_S11_GMFA.fdt', '._52SC_V1_120SP_M1_Merged_S11_GMFA.fdt', '52SC_V1_120SP_M1_Merged_S11_GMFA.set', '._52SC_V1_120SP_M1_Merged_S11_GMFA.set', '53SH_V3_120SP_M1_Merged_S10_Final.fdt', '._53SH_V3_120SP_M1_Merged_S10_Final.fdt', '53SH_V3_120SP_M1_Merged_S10_Final.set', '._53SH_V3_120SP_M1_Merged_S10_Final.set', '56YM_V2_120SP_M1_Merged_S11_GMFA.fdt', '._56YM_V2_120SP_M1_Merged_S11_GMFA.fdt', '56YM_V2_120SP_M1_Merged_S11_GMFA.set', '._56YM_V2_120SP_M1_Merged_S11_GMFA.set', '57AH_V2_120SP_M1_Merged_S10_Final.fdt', ]     

The error is different because in your  script you put the .fdt file instead of .set file into the mne.io.read_epochs_eeglab function. Otherwise the error is the same. 

Thanks

drammock, 2020-01-10:
OK, thanks.  Sorry, but we had to rule out user error first.  But it seems this might be a bug.  I don't normally work with EEGLAB files so I'm pinging the other devs on [our gitter channel](https://gitter.im/mne-tools/mne-python) to see if one of them has seen this before.

agramfort, 2020-01-11:
please a short script and the files that are problematic so we can
investigate

>

Davi1990, 2020-01-11:
Dear Alex, 

please find below the link to download both the .fdt and .set file. 

https://www.dropbox.com/s/0n8zs7t5gobldwu/80LP_V1_SP_M1_S10_Final.fdt?dl=0

https://www.dropbox.com/s/j5kbbyiecrju8rj/80LP_V1_SP_M1_S10_Final.set?dl=0

The files correspond to one among the problematic subjects and it is the same of the previous messages. 

Thanks

mmagnuski, 2020-01-11:
@Davi1990 I'm looking into the files now, I will let you know what I find.

mmagnuski, 2020-01-11:
@Davi1990 @agramfort @drammock 
Ok, the error is related to the `EEG.data` value, which is different to the file name. Notice that the error says:
```python-traceback
Traceback (most recent call last):

  File "<ipython-input-24-da692532ba90>", line 1, in <module>
    epochs = mne.io.read_epochs_eeglab(op.join(data_dir, '80LP_V1_SP_M1_S10_Final.set'))

  File "c:\src\mne-python\mne\io\eeglab\eeglab.py", line 243, in read_epochs_eeglab
    uint16_codec=uint16_codec)

  File "<c:\src\mne-python\mne\externals\decorator.py:decorator-gen-195>", line 2, in __init__

  File "c:\src\mne-python\mne\utils\_logging.py", line 90, in wrapper
    return function(*args, **kwargs)

  File "c:\src\mne-python\mne\io\eeglab\eeglab.py", line 549, in __init__
    with open(data_fname, 'rb') as data_fid:

FileNotFoundError: [Errno 2] No such file or directory: 'C:\\tmp\\test_mne_7192\\80_LP_V1_SP_M1_S10_Final.fdt'
```
while the file name in the directory is `80LP_V1_SP_M1_S10_Final.fdt` (so there is `80LP`, not `80_LP` at the beginning).
There are two quick solutions on your end:
* you can change the file name to agree with the `EEG.data` value
* or you can edit the `EEG.data` to have the value that is the same to the actual file name.

On the mne side, because it is not the first time this error appears, I think we might change how fdt files are read: when the `EEG.data` filename is absent and it is different than `fdt_fname = set_fname.replace('.set', '.fdt')` then we could use `fdt_fname`.
This problem should be common when users rename the `.set` and `.fdt` files by hand but not through eeglab (which also changes the `EEG.data` value when saving under a different name). @Davi1990 - are you able to read the file in EEGLAB? I think it does the same that I am proposing here.

Davi1990, 2020-01-11:
@mmagnuski thanks for your update. I've changed the name in MATLAB so that now everything match. Of course, now I was able to run ` mne.io.read_epochs_eeglab`with no issue. 

I am analyzing the same dataset in MATLAB for some convolutional stuffs but I've never faced this problem before. 

Thanks again for your help...really appreciate that!

agramfort, 2020-01-11:
I  confirm that EEGLAB can read these files but raises a warning:

Warning: the name of the dataset has changed on disk, updating .dat & .fdt
data file to the new name
Reading float file '/Users/alex/Downloads/80LP_V1_SP_M1_S10_Final.fdt'...

+1 for this the behavior of MNE to match eeglab

>

mmagnuski, 2020-01-11:
Ok, I started doing that, I'll submit a PR later.

PR #7194 (2020-01-15): Eeglab reader: look for fdt if EEG.data file name is wrong
#### Reference issue
Fixes #7192

#### What does this implement/fix?
When `EEG.data` file name is wrong (the file is not present) an additional check is performed: set file name is used replacing ".set" with ".fdt". This is a common problem when set files are renamed in the operating system and not via eeglab (which also changes `EEG.data`).
closed 2020-01-15 by agramfort