mne-tools/mne-python #7195: bem._prepare_env()'s requires_freesurfer arg

hoechenberger, 2020-01-12:
Commit https://github.com/mne-tools/mne-python/commit/799c2ecb74604ad0ab432b6cee5739dca5c85ca6 removed the use of the `requires_freesurfer` parameter inside `bem._prepare_env()`; yet, it's still a required argument of the function itself. I wonder if that was intentional (is so, what was the reasoning?) or @larsoner just overlooked something there?

agramfort, 2020-01-12:
maybe it's required by the public function calling this private one so it
was redundant?

>

larsoner, 2020-01-12:
It looks like `require_freesurfer=True` is always set. I probably meant to remove it in the function and in the calls to the function but forgot.

hoechenberger, 2020-01-12:
I can probably it in #7196 too

closed 2020-01-13 by larsoner