mne-tools/mne-python #7203: IRASA method

palday, 2020-01-13:
Quoting from the [FieldTrip website](http://www.fieldtriptoolbox.org/example/irasa/)

>IRASA allows distinguishing rhythmic activity from concurrent power-spectral 1/f modulations. The technique virtually compresses and expands the time-domain data with a set of non-integer resampling factors prior to Fourier-based spectral decomposition. As a result, rhythmic components in the power-spectrum are redistributed while the arrhythmic 1/f distribution is left intact. Taking the median of the resulting auto-spectral distributions extracts the power-spectral 1/f component, and the subsequent removal of the 1/f component from the original power-spectrum offers a power-spectral estimate of rhythmic content in the recorded signal.

The [math on this doesn't appear too bad]( https://doi.org/10.1007/s10548-015-0448-0), but figuring out an appropriate interface might be a little bit more complicated, including what to return and which classes to have this as a method.

raphaelvallat, 2020-01-13:
In case that's helpful, I've implemented IRASA (as a function) here:
https://github.com/raphaelvallat/yasa/blob/200659872405fcc7d5477fd1069e328a372d3e9b/yasa/spectral.py#L229

palday, 2020-01-14:
That's quite helpful! Especially since it's already licensed under 3 clause BSD.

larsoner, 2020-01-14:
I guess this would be an `mne.preprocessing` method?

@palday do you want to try the method out using @raphaelvallat's code first?

palday, 2020-01-14:
That's the plan, but it definitely won't happen this week and probably not next....