mne-tools/mne-python #7204: Plotting/Visualisation Issue with Montage/MNE

DarrenRhodes, 2020-01-13:
Dear MNE team

Forgive me first for being a GutHub newbie and an MNE newbie.

I have managed to load in and work with some Biosemi128 data. However I am running into some issue regarding visualizing some of the output.

I have defined the montage as:

>>>ten_twenty_montage = mne.channels.make_standard_montage('biosemi128')

(I understand I need to update the variable name).

However, when it comes to plotting down the pipeline, I have done this:

>>>Stim3_epochs.plot_topomap(times=[0., 0.08, 0.1, 0.12, 0.2], ch_type='eeg')

(Stim3 is just a condition in my experimental data - nothing interesting). 

I get the following error:

ValueError: The following electrodes have overlapping positions, which causes problems during visualization:
EXG1, EXG2, EXG3, EXG4, EXG5, EXG6, EXG7, EXG8, GSR1, GSR2, Erg1, Erg2, Resp, Plet, Temp

I imagine there is some way to exclude or combine these but I am very unsure...

Sorry in advance if it's a silly question/issue.

Darren

agramfort, 2020-01-13:
can you try to make this electrodes as MISC and not EEG with
https://mne.tools/dev/generated/mne.io.Raw.html#mne.io.Raw.set_channel_types
 ?

>

DarrenRhodes, 2020-01-13:
Yep, works perfect! Brilliant. Thank you so much.

larsoner, 2020-01-13:
@DarrenRhodes do you want to make a PR to improve the error message here to suggest using `inst.set_channel_types` if the channels are not M/EEG channels?

https://github.com/mne-tools/mne-python/blob/master/mne/channels/layout.py#L712-L714

closed 2020-01-13 by larsoner