mne-tools/mne-python #7208: mne.read_events can't read events from MNE-BIDS events file

hyruuk, 2020-01-14:
Hi !

I converted my data to BIDS format from the CTF MEG format (.ds) using MNE-BIDS. That created a set of files including {BIDS-structure}_events.tsv files for each recording. I now want to load the events into an MNE events array. However, the mne.read_events() function doesn't seem to work on these files. When called, it returns the following error : 

`</home/karim/electrophy/lib/python3.6/site-packages/mne/externals/decorator.py:decorator-gen-135> in read_events(filename, include, exclude, mask, mask_type, verbose)

~/electrophy/lib/python3.6/site-packages/mne/utils/_logging.py in wrapper(*args, **kwargs)
     88             with use_log_level(verbose_level):
     89                 return function(*args, **kwargs)
---> 90         return function(*args, **kwargs)
     91     return FunctionMaker.create(
     92         function, 'return decfunc(%(signature)s)',

~/electrophy/lib/python3.6/site-packages/mne/event.py in read_events(filename, include, exclude, mask, mask_type, verbose)
    254         #  Have to read this in as float64 then convert because old style
    255         #  eve/lst files had a second float column that will raise errors
--> 256         lines = np.loadtxt(filename, dtype=np.float64).astype(int)
    257         if len(lines) == 0:
    258             raise ValueError('No text lines found')

~/electrophy/lib/python3.6/site-packages/numpy/lib/npyio.py in loadtxt(fname, dtype, comments, delimiter, converters, skiprows, usecols, unpack, ndmin, encoding)
   1099         # converting the data
   1100         X = None
-> 1101         for x in read_data(_loadtxt_chunksize):
   1102             if X is None:
   1103                 X = np.array(x, dtype)

~/electrophy/lib/python3.6/site-packages/numpy/lib/npyio.py in read_data(chunk_size)
   1026 
   1027             # Convert each value according to its column and store
-> 1028             items = [conv(val) for (conv, val) in zip(converters, vals)]
   1029 
   1030             # Then pack it according to the dtype's nesting

~/electrophy/lib/python3.6/site-packages/numpy/lib/npyio.py in <listcomp>(.0)
   1026 
   1027             # Convert each value according to its column and store
-> 1028             items = [conv(val) for (conv, val) in zip(converters, vals)]
   1029 
   1030             # Then pack it according to the dtype's nesting

~/electrophy/lib/python3.6/site-packages/numpy/lib/npyio.py in floatconv(x)
    744         if '0x' in x:
    745             return float.fromhex(x)
--> 746         return float(x)
    747 
    748     typ = dtype.type

ValueError: could not convert string to float: 'onset'
`

I assume it's because of the headers in the tsv formatted file, but I would expect this function to be able to read these.
Is there a way to easily load these files into MNE events format (without using find_events on the raw data file) ?

agramfort, 2020-01-15:
mne.read_events is to read MNE events which are fif files or basic txt
files and not BIDS tsv files which were standardized years after MNE
formats.

for reading bids data use mne-bids

>

closed 2020-01-15 by agramfort

jasmainak, 2020-01-18:
Please take a look at this example:

https://mne.tools/mne-bids/auto_examples/convert_mne_sample.html#sphx-glr-auto-examples-convert-mne-sample-py

It's not very discoverable at the moment. We are working on creating a separate example for this.

Immiora, 2021-02-10:
> Please take a look at this example:
> 
> https://mne.tools/mne-bids/auto_examples/convert_mne_sample.html#sphx-glr-auto-examples-convert-mne-sample-py
> 
> It's not very discoverable at the moment. We are working on creating a separate example for this.

Unfortunately, the link does not seem to work anymore. Was the example moved somewhere else?

Immiora, 2021-02-10:
I supposed it is now here: https://github.com/mne-tools/mne-bids/blob/master/examples/convert_mne_sample.py

drammock, 2021-02-10:
Rendered version is here:

https://mne.tools/mne-bids/stable/auto_examples/convert_mne_sample.html