mne-tools/mne-python #7213: BUG: Viz Topomap Produces ValueError Always Due to Cardinal Dig

adam2392, 2020-01-15:
## Describe the bug

In `mne.viz.topomap(data, info)`, I get an error in the latest version:

```
ValueError: Number of channels in the Info object and the data array does not match. Pick Info with e.g. mne.pick_info and mne.io.pick.channel_indices_by_type.
```

#### Steps to reproduce
If I run the following code:

```
from pprint import pprint

montage = mne.channels.make_standard_montage("standard_1020")
raw.set_montage(montage)
info = raw.info
pprint(info['dig'])
```

You'll notice that we have extra DigPoints.

#### Expected results
It should work as before (I think ~v0.19?) where it just plotted the topomap as long as the data shape matched.

If you use `mne.viz.plot_sensors`, it works as expected and only plots the EEG sensors.

larsoner, 2020-01-15:
@adam2392 this is not enough for us to reproduce the bug locally. Can you include an example that we can copy-paste to see it locally? For example you can create a dummy `Raw` with something like:
```Python
import numpy as np
import mne
data = np.random.RandomState(0).randn(2, 1000)
info = mne.create_info(['Cz', 'Fz'], 1000., 'eeg')
raw = mne.io.RawArray(data, info)
montage = mne.channels.make_standard_montage("standard_1020")
raw.set_montage(montage)
```
This does not crash. Can you adapt it to your use case so that it does?

Alternatively you can upload the raw data somewhere and give us a link.

adam2392, 2020-01-15:
Ah okay, so the issue arises only when there is a channel marked in the `bads` list. If you have 20 EEG channels total, and one is marked in the `bads` list, and you pass in a vector 20x1, then it breaks.

I suppose in a way this would "be expected", but would it be beneficial to make the error possibly more informative?

larsoner, 2020-01-15:
> Ah okay, so the issue arises only when there is a channel marked in the bads list. If you have 20 EEG channels total, and one is marked in the bads list, and you pass in a vector 20x1, then it breaks.

I'm not quite following. Can you modify my code snippet above to produce the breaking behavior? That will help me understand where you're hitting the error and what the situation is (where bads come in, data sizes, channels, etc.).

mmagnuski, 2020-01-15:
@adam2392 - you are correct, @larsoner see the snippet below:
```python
import numpy as np
import mne

ch_names = list('abcde')
ch_pos = np.array([[0, 0, 1], [1, 0, 0], [-1, 0, 0],
                   [0, -1, 0], [0, 1, 0]])
ch_pos_dict = {name: pos for name, pos in zip(ch_names, ch_pos)}
dig = mne.channels.make_dig_montage(ch_pos_dict, coord_frame='head')

data = np.random.random(5)
info = mne.create_info(ch_names, 250, ['eeg'] * 5)
info.set_montage(dig)
info['bads'] = ['a']

ax, _ = mne.viz.plot_topomap(data, info)
```

I'm not sure whether this should be expected. It is kind of surprising, but its not unreasonable. I don't have a strong opinion here.

larsoner, 2020-01-16:
Thanks for the reproducible example, @mmagnuski @adam2392 see #7220

PR #7220 (2020-01-18): FIX: Fix plot_topomap with bads
Closes #7213
closed 2020-01-18 by agramfort