mne-tools/mne-python #7216: Default brainmask name in mne.bem.make_watershed_bem would incur error by Freesurfer(dev) mri_watershed

zuxfoucault, 2020-01-16:
#### Describe the bug
Default brainmask name in `mne.bem.make_watershed_bem` would incur error by Freesurfer (dev a76846c) `mri_watershed`

https://github.com/mne-tools/mne-python/blob/maint/0.19/mne/bem.py#L1036

******************************
Saving /home/foucault/data/derivatives/freesurfer/sub-103/bem/watershed/ws
error: unknown file type for file (/home/foucault/data/derivatives/freesurfer/sub103/bem/watershed/ws)

If the `brainmask='brainmask.mgz'` is set, there's no error message.

agramfort, 2020-01-16:
what is the fix?

>

zuxfoucault, 2020-01-16:
Replace the default value `brainmask='ws'` with `brainmask='brainmask.mgz'`?

Based on the documentation: https://github.com/mne-tools/mne-python/blame/master/doc/_includes/bem_model.rst#L42
I guess the original design intended to set `ws` as a folder name, but `ws` here is just a file name. Perhaps, a structure like `bem/watershed/brainmask.mgz` would be satisfied?

The output folder structure after after modifacation of the default brainmask value to `brainmask.mgz` would like :
.
├── brain.surf -> watershed/sub-103_brain_surface
├── inner_skull.surf -> watershed/sub-103_inner_skull_surface
├── outer_skin.surf -> watershed/sub-103_outer_skin_surface
├── outer_skull.surf -> watershed/sub-103_outer_skull_surface
├── sub-103-head.fif
└── watershed
    ├── brainmask.mgz
    ├── sub-103_brain_surface
    ├── sub-103_inner_skull_surface
    ├── sub-103_outer_skin_surface 

I could do a PR for this if it is appropriate.

larsoner, 2020-01-16:
That sounds reasonable to me

zuxfoucault, 2020-01-17:
In the latest master (01b90ce), I couldn't see the stdout. How to make stdout and stderr printed?
The parameters and function called:
`mne.bem.make_watershed_bem(subject, subjects_dir=SUBJECTS_DIR, overwrite=True, brainmask='ws.mgz', verbose='debug')`

larsoner, 2020-01-17:
`verbose=True` and especially `verbose='debug'` should probably be enough to see the output printed. Might be a bug in `master` somewhere. But in the meantime just for testing you should be able to hack any `run_subprocess` calls to do:
```
output = run_subproces(...)
print(output)
```
or so.

zuxfoucault, 2020-01-17:
It's wrapped by partial, uh, how to print it?
```
def make_watershed_bem(subject, subjects_dir=None, ...

       run_subprocess_env = partial(run_subprocess, env=env, cwd=tempdir)
```
I also tried `print(run_subprocess_env)`, nothing showed.

PR #7223 (2020-01-18): MRG, FIX, DOC: modify brainmask default value
#### Reference issue
Fixes #7216

#### What does this implement/fix?
Modify the brainmask default value in `mne.bem.make_watershed_bem` and the corresponding part in the document.
closed 2020-01-18 by agramfort