mne-tools/mne-python #7221: Orientation in plot_bem

christian-oreilly, 2020-01-16:
#### Describe the bug
Two things. The first one is more of a feature request: It might be nice to have an optional mri arguments to mne.viz.plot_bem to explicitly pass the location of the MRI file to use in mne.viz.plot_bem.

Second, in the current code of MNE, the orientation argument of mne.viz.plot_bem only produce correct visualization when the MRI volumes are in the "conform" space (e.g, running  mri_convert with --conform). E.g., here is the result with orientation == coronal for a volume that is not in the "conform" space:

[image]

To the extent that the volumes contain the metadata for the orientation, the plot_bem function should check the orientation metadata and either 1) throw an exception if using non-conform volumes would cause to many issues or 2) give an appropriate view of the volume if managing internally non-conform volumes is reasonable for MNE. I don't think the volume should be made "conform" by MNE, unless MNE also manage rotating all the other artifacts (e.g., BEM surfaces) so that they stay in registration with the volume.

Question: Are there other functions that would be impacted by non "conform" volume?

christian-oreilly, 2020-01-16:
@larsoner As discussed on the mailing list.

larsoner, 2020-01-16:
> To the extent that the volumes contain the metadata for the orientation, the plot_bem function should check the orientation metadata and either 1) throw an exception if using non-conform volumes [...]

The easiest (well, simplest?) fix is to use [nib.io_orientation](https://nipy.org/nibabel/reference/nibabel.orientations.html#nibabel.orientations.io_orientation) or [nib.aff2axcodes](https://nipy.org/nibabel/reference/nibabel.orientations.html#nibabel.orientations.aff2axcodes) to "just" modify our MRI slicing and create the appropriate reordering/negation affine and apply it to the surface `rr` values. @christian-oreilly interested in trying?

Another option would be to use:

https://nipy.org/nibabel/reference/nibabel.funcs.html#nibabel.funcs.as_closest_canonical

to get an updated image, then use linear algebra between the original affine and the closest-canonical one (it's going to be a dot product between one and the inverse of the other, which should yield a matrix with three non-zero entries all either +1 or -1) to figure out the transformation to apply to the surfaces for display. But this seems like more work for something that would end up being equivalent.

> Are there other functions that would be impacted by non "conform" volume?

Other than `plot_bem`, I would look at `plot_alignment` of your data. This is designed to use the same transformation paths as many other things in MNE (e.g., BEM calculations, forward solution, etc.).

At least `plot_volume_source_estimates` already makes use of `aff2axcodes` to figure out data orientation for clicking and so forth.

Once you go to visualize with PySurfer, it probably will assume your MRI is oriented the standard way and so, e.g., when you ask for a lateral view you will not get it. I think I worked with some data in non-RAS orientation once and had that problem. In theory this could be fixable / should be fixed in PySurfer the same way as above.

larsoner, 2020-04-27:
@jasmainak since you have worked a bit on the 3D plotting maybe you'd be interested in tackling this? I think the `io_orientation` or `aff2axcodes` would be the way to go for simplicity and speed.

jasmainak, 2020-05-03:
i'm sorry with the late response. How urgent is this? The original request seems straightforward to fix but the rest might need work. I am a bit swamped at the moment due to an upcoming grant deadline, but I can file it in my todos and would be happy to give it a stab after that.

PR #7725 (2020-05-11): MRG, FIX: Fix MRI orientations
Closes #7221.

@christian-oreilly can you try it with your MRIs that are in non-standard orientation? I was able to check that the FreeSurfer-standard ones look okay, but don't have good MRIs to test for non-standard orientations. If you don't have them I could maybe resample these and regenerate BEMs but that sounds like too much work :)

This actually makes our code simpler and easier to read I think, which is nice.
closed 2020-05-11 by agramfort