mne-tools/mne-python #7225: Beamformer: Selection of orientation that optimizes output power.

britta-wstnr, 2020-01-17:
I am sitting with @wmvanvliet and @ckiefer0 and we noticed that there are two points in the beamformer code where we compute the optimal orientation `pick_ori='max-power'` and that those two snippets do not give the same weights if copied into the same place.
Specifically, the optimal orientation gets computed differently based on whether `weight_norm='unit-noise-gain'` or `weight_norm=None`.
@wmvanvliet looked deeper into it and saw that the orientation is not recovered very well for the no weight normalization version.

The difference between the versions is whether the orientation gets picked before the computation of the weights (`weight_norm='unit-noise-gain'`) or after (`weight_norm=None`).

We are still evaluating the issue, although it looks like the unit-noise-gain case is the right solution. We suggest to refactor the code such that both branches reach the same orientation optimization code.

Code references:
`weight_norm=unit-noise-gain`:
https://github.com/mne-tools/mne-python/blob/master/mne/beamformer/_compute_beamformer.py#L157

`weight_norm=None`:
https://github.com/mne-tools/mne-python/blob/master/mne/beamformer/_compute_beamformer.py#L289

agramfort, 2020-01-17:
sounds good. A simulation tells you what version is correct. thanks for
looking into it

>

britta-wstnr, 2020-01-17:
@agramfort simulation results are underway, Marijn is onto it!

larsoner, 2020-01-17:
Awesome @britta-wstnr . Do you have time to also look at #7168 while you're at it?

britta-wstnr, 2020-01-17:
@larsoner oh I missed that one, apologies. I will look at it!

wmvanvliet, 2020-01-18:
Simulation results are up! Now with an additional column: orientation error.

https://users.aalto.fi/~vanvlm1/beamformer_simulation/lcmv_vol.html

Seems that both code paths *can* produce more or less reliable orientation estimates, but it depends on the `reduce_rank` parameter.

agramfort, 2020-01-19:
that's a lot of results to inspect. Any particular summary / take home
message?

>

britta-wstnr, 2020-01-19:
@wmvanvliet the color bar is hard to read, but I guess brighter colours are better = less error? If I see it right, then the `weight_norm='unit-noise-gain'` code path seems to produce stable results more reliably, although, it too can fail.
If I would refactor the code in favour of that code path, would you be able to see with the simulation whether the `weight_norm=None` path gets more reliable outcomes (i.e., gets better and not worse)?

wmvanvliet, 2020-01-19:
@britta-wstnr I agree with your analysis. And yes, if you refactor, I can do a side-by-side comparison.

@agramfort take home message: one code path does not necessarily produce worse orientation estimations that the other. It depends on the other parameters. Notably `reduce_rank`. Still,one of the code paths seems to produce better results more often than the other path, hence might be preferable. In the end, it seems more elegant to me to have a single code path and that should be possible.

agramfort, 2020-01-20:
whatever you think is best :)

>

britta-wstnr, 2020-01-20:
I'll get to it then!

larsoner, 2023-10-01:
I think this was fixed with #7656 so I'll close

closed 2023-10-01 by larsoner