mne-tools/mne-python #7228: BUG: Bug with multi-view screen capture

larsoner, 2020-01-17:
See this example that uses `hemi='split'`, all captured PyVista images are duplicated:

https://mne.tools/dev/auto_examples/visualization/plot_publication_figure.html#sphx-glr-auto-examples-visualization-plot-publication-figure-py

Moreover, once `brain.close()` is called, it should make sure no figures are generated (as they have been closed for that snippet). Compare the above to what used to be produced with `mayavi` produced (one pair of images rather than four pairs):

https://mne.tools/stable/auto_examples/visualization/plot_publication_figure.html#sphx-glr-auto-examples-visualization-plot-publication-figure-py

cc @GuillaumeFavelier can you look into it?

GuillaumeFavelier, 2020-01-17:
Yep, I'll take care of it.

GuillaumeFavelier, 2020-01-20:
I reported this behaviour upstream: https://github.com/pyvista/pyvista/pull/548

GuillaumeFavelier, 2020-02-14:
Unfortunately, I think this might take a while to fix, probably at the next release of PyVista

larsoner, 2020-03-04:
@GuillaumeFavelier it would be nice not to have our docs have this problem when they go to `stable`. So I think we either need this for 0.20, or we temporarily revert to using Mayavi to build docs during release...

larsoner, 2020-03-05:
@GuillaumeFavelier should we revert to `Mayavi` on `CircleCI` until this is fixed?

GuillaumeFavelier, 2020-03-06:
> should we revert to Mayavi on CircleCI until this is fixed?

+1 for this @larsoner, I'll open the PR

PR #7416 (2020-03-10): MRG, MAINT: Work around PyVista rendering bug
Workaround for PyVista scraping. @GuillaumeFavelier feel free to look and decide if it needs a backport (not sure what the PyVista scraper duplication status is). All I really did was add the `if plotter in seen_plotters:` check and accumulation and it appears to fix things.

Alternative to #7398
Closes #7398
Closes #7228
closed 2020-03-10 by GuillaumeFavelier