mne-tools/mne-python #7229: mne.find_events returning no stim channel found

danushaaditya, 2020-01-18:


agramfort, 2020-01-18:
can you be more explicit? for data that have no stim channels events are
now available in raw.annotations and you can get them with

events, _ = mne.events_from_annotations(raw)

does it fix the issue?

>

danushaaditya, 2020-01-18:
I tried it, now it shows a new error.
AttributeError: module 'mne' has no attribute 'event_from_annotations'

I'm new to python so i'm not sure what i am doing wrong.
Thanks for helping out

drammock, 2020-01-18:
You are running an example from version 0.14 of the documentation (look at the URL). Current version of MNE-Python is 0.19. At the very least, make sure your installed version matches the version of the docs you are working from. We also strongly recommend updating to the newest version.

agramfort, 2020-01-18:
https://mne.tools/stable/generated/mne.events_from_annotations.html

events with an s

>

danushaaditya, 2020-01-19:
@agramfort Thank you. The 's' was missing. I was being careless.
Thanks a lot for your help.

closed 2020-01-19 by danushaaditya