mne-tools/mne-python #7230: spectral_connectivity( )  issue

Charlesi8, 2020-01-19:
hi, guys , i have a puzzle abount function spectral_connectivity( ).When i run this function on my own epochs data with the parameters like 
cwt_freqs=np.arange(8,13,0.5)
cwt_n_cycles=1  con1,freqs1,times1,n_epochs1=spectral_connectivity(data['button_tone'],method='coh',mode='cwt_morlet',sfreq=1024, cwt_freqs=cwt_freqs,cwt_n_cycles=cwt_n_cycles, n_jobs=1)
, an error happend with words like " **ValueError: too many values to unpack (expected 4)**"
And i do not know how to deal with it , could some one help me ,plz. 
[image]

agramfort, 2020-01-19:
can you share a code snippet to replicate?

>

Charlesi8, 2020-01-20:
```
data=mne.read_epochs(x)     # x is a *.fif file
data1=data['button_tone']   
#'button_tone 'is one of the conditions ; the shape of data1 is like (100,65,3072)  ( n_epochs,n_channels,n_samples)

cwt_freqs=np.arange(8,13,0.5)
cwt_n_cycles=1
con,freqs,times,n_epochs=spectral_connectivity(data1,method='coh',mode='cwt_morlet',sfreq=1024, cwt_freqs=cwt_freqs,cwt_n_cycles=cwt_n_cycles, n_jobs=1)

```
And one error happened as i said above

agramfort, 2020-01-20:
if I don't have x I cannot replicate. You can make fake dataset with
mne.EpochsArray. Please make sure I can replicate the pb on my machine with
just a copy-paste

>

mmagnuski, 2020-01-20:
@Charlesi8 Look at the function docs: https://mne.tools/stable/generated/mne.connectivity.spectral_connectivity.html
you are expecting 4 output arguments, but the function returns 5. The error is not in mne, but in your code. You can just do:
```python
con1, freqs1, times1, n_epochs, _ = spectral_connectivity(
    data['button_tone'], method='coh', mode='cwt_morlet', sfreq=1024, 
    cwt_freqs=cwt_freqs, cwt_n_cycles=cwt_n_cycles, n_jobs=1)
```

closed 2020-01-20 by mmagnuski

Charlesi8, 2020-01-21:
i have revised my code, and it really helped .Thank you so much.

mmagnuski, 2020-01-21:
@Charlesi8 glad to help!