mne-tools/mne-python #7234: bug brain.add_label when hemi='split'

agramfort, 2020-01-21:
trying this 

```
brain = stc.plot(subject='sample', time_viewer=True, hemi='split', views=['lat', 'med'])
brain.add_label('~/mne_data/MNE-sample-data/subjects/sample/label/lh.BA1.label')
```

I get a non meaningful error. It should work though as I know it's a left hemi label so it should be added only to left hemi.

cc @GuillaumeFavelier

GuillaumeFavelier, 2020-01-23:
I think this is just not supported yet, looking at the doc of `add_label()`:

```
        hemi : str | None
            If None, it is assumed to belong to the hemipshere being
            shown. If two hemispheres are being shown, an error will
            be thrown.
```

We could work to integrate it into `_Brain` though :+1: 

> I know it's a left hemi label 

Can we programmatically get this information?

agramfort, 2020-01-23:
with read_label you get a label that should have an hemi attribute
https://github.com/mne-tools/mne-python/blob/master/mne/label.py#L233 to
tell you

>

GuillaumeFavelier, 2020-01-23:
This is great! I'll work on someting then

GuillaumeFavelier, 2020-01-23:
Meanwhile could you tell me if the following works for you?

```py
    brain = stc.plot(subject='sample', hemi='split', views=['lat', 'med'])
    label = read_label('~/mne_data/MNE-sample-data/subjects/sample/label/lh.BA1.label')
    brain.add_label(label)
```

PR #7242 (2020-01-23): Allow label=filepath in add_label for _Brain
This PR delays `_check_hemi()` to allow the user to pass `label=filepath`.

Closes #7234
closed 2020-01-23 by larsoner