mne-tools/mne-python #7235: picking vertex in brain object to see time course

agramfort, 2020-01-21:
I would be great to get a plot of a time course of a vertex on a 3D brain object.

a suggestion from @SophieHerbst is also to have a short cut to pick automatically the vertex with the highest amplitude in the current view.

larsoner, 2020-01-21:
> I would be great to get a plot of a time course of a vertex on a 3D brain object.

At least in principle there is vtkChart:

- https://vtk.org/doc/nightly/html/classvtkChart.html
- https://vtk.org/Wiki/VTK/Examples/Cxx/Plotting/LinePlot

> a suggestion from @SophieHerbst is also to have a short cut to pick automatically the vertex with the highest amplitude in the current view.

I'm actually not sure the extent to which this will be possible. Picking with the mouse is usually done by rendering tricks or casting a ray (I think VTK [does the latter](https://vtk.org/doc/nightly/html/classvtkPicker.html)). I'm not sure there is a way to ask for "all mesh vertices currently visible). And given a non-zero alpha, this isn't even necessarily well defined. So for a first pass I say we just make it so that clicking on a point in the mesh does mesh vertex picking and does a vtkChart plot of the time course. (If off the mesh, don't change the point currently used. This way clicking always will show the time course, and if you don't want it to change, click in the empty space around the mesh. Either that or use ctrl-click for picking or something.)

I leave it as an exercise to @GuillaumeFavelier to figure out where this plot should live :) Maybe a `show_timecourse=True` would make a new full-width (regardless of `views`) plot area at the bottom reserved for a time course chart like our volume viewer:

https://mne.tools/dev/auto_examples/inverse/plot_lcmv_beamformer_volume.html#plot-source-space-activity

jasmainak, 2020-01-21:
why not extend the functionality of 4D viewer with cortical ribbon first?

larsoner, 2020-01-21:
It's nicer to view surface source estimate data on the surface

GuillaumeFavelier, 2020-01-21:
That's a nice idea and thank you so much for the pointers @larsoner, I have somewhere to start now.

larsoner, 2020-01-21:
And for mouse click vertex picking in mayavi (so you can translate to PyVista) see

https://github.com/mne-tools/mne-python/blob/master/mne/gui/_fiducials_gui.py#L359

GuillaumeFavelier, 2020-01-23:
[Vertex picking is available](https://docs.pyvista.org/plotting/plotting.html?highlight=enable_point_picking#pyvista.BasePlotter.enable_point_picking) by default in PyVista:

[image]

SophieHerbst, 2020-01-23:
Would this allow to spit out MNI coordinates at the pick (or the label of a pre-defined atlas)?

SophieHerbst, 2020-01-23:
Also, a very small add-on labels for the 3 clim-scales would be helpful.

GuillaumeFavelier, 2020-01-23:
I think any data associated to the point data/vertex id could be retrieved that way in theory.

GuillaumeFavelier, 2020-01-23:
I discovered [`vertex_to_mni()`](https://mne.tools/stable/generated/mne.vertex_to_mni.html#mne.vertex_to_mni) to take care of the conversion.

GuillaumeFavelier, 2020-01-23:
> Also, a very small add-on: labels for the 3 clim-scales would be helpful.

fmax/fmid/fmin labels were present [in previous version](https://github.com/mne-tools/mne-python/pull/7188) but were dropped in favour of a more compact UI.

GuillaumeFavelier, 2020-01-24:
**ToDo** (suggested in https://github.com/mne-tools/mne-python/pull/7247#issuecomment-577954381)

- [ ] Enable Picking & Plot of time courses with `show_traces=True`
- [ ] Display the picked point in 3d with a dot glyph (sphere)
- [ ] Add a time course to the chart by picking on the suface vertex
- [ ] Remove an existing time course in the chart by clicking on the dot glyph
- [ ] Display the chart with vtkChartXY in split view

larsoner, 2020-01-27:
+10 to embedding a mpl figure rather than figuring out vtkChartXY, it gives us more flexibility going forward, e.g., including showing time-frequency images eventually, and will unify the user experience of 2D plots around MPL and its standard interactions (like zooming and view resetting). The "only" tricky parts of this approach will be:

- setting up our own Qt window instead of relying on [PyVista](https://docs.pyvista.org/plotting/qt_plotting.html) and [MPL](https://matplotlib.org/gallery/user_interfaces/embedding_in_qt5_sgskip.html) to do it for us, and then also
- eventually making sure that when in a notebook we embed the canvases properly rather than creating a Qt window

mmagnuski, 2020-01-28:
@larsoner Why would we have to set up our own Qt window? At least in interactive matplotlib mode - it might be easier and more flexible to open a separate matplotlib window per click/interaction.

larsoner, 2020-01-28:
I don't think that would work as well. The design I'm advocating is basically a variant of what we already do in the volume plotter. You could apply your argument to that as well -- that we shouldn't render the time trace(s) in the same window as the brain volume rendering -- and I don't think it would be as nice as experience as having them contained in the same window.

mmagnuski, 2020-01-28:
Yes, that's true - separate window can feel more clunky, but the benefit is that the user can place it however he wants. And although volume source plotter shows traces in the same window - many other mne interfaces open separate windows (time-frequency - when interacting with topomap [also tfr heatmap IIRC], evokeds - when selecting time window, ica source signal viewer when clicking on ic label, ica plot components opens properties windows). With separate windows you can also compare various traces without the need to implement specific option for that - you click on two vertices / channels and see corresponding signals in two separate windows.
In general I agree that it might be nicer to have everyting in one plot - my previous comment was mostly motivated by perceived difficulity of this goal. It seems easier to spawn a separate matplotlib window. But the one-window approach may be easier than I think - and the user experience may be worth it. :)

GuillaumeFavelier, 2020-01-29:
> my previous comment was mostly motivated by perceived difficulity of this goal.

Honestly I thought the same at first but it seems only [a few lines of code](https://github.com/mne-tools/mne-python/blob/28cc5e33c97aa5cdce45a1b2d91d3a4ecb192d4f/mne/viz/_brain/_timeviewer.py#L402-L405) were necessary to add this:

```py
        vlayout = self.plotter.frame.layout()
        vlayout.addWidget(self.mpl_canvas)
        vlayout.setStretch(0, 2)
        vlayout.setStretch(1, 1)
```

mmagnuski, 2020-01-29:
@GuillaumeFavelier - that's cool!
So maybe it wouldn't be difficult to add an option to detach the matplotlib figure? 😈

GuillaumeFavelier, 2020-01-29:
It's possible too. How would you pass this option though? We already use `show_traces=True` in `plot_source_estimates()`.

larsoner, 2020-01-29:
`show_traces='separate'` or something. But let's figure this API out separately as there are additional questions we'll want to sort out (e.g., should each click open a new window, should we close the windows when clicking just like we remove traces from the all-one plot in the joint mode, etc.)

mmagnuski, 2020-01-29:
@larsoner Sure!

PR #7247 (2020-02-27): MRG: Enable point picking in _TimeViewer
This PR enables point picking and display the vertex id as text in the top left. It's the first step to achieve #7235. Shortcut is `p` to pick under the mouse. This is still a work in progress.

**Animation**

[image]

**ToDo**

https://github.com/mne-tools/mne-python/issues/7235#issuecomment-577959594

Closes #7235 
It's an item of https://github.com/mne-tools/mne-python/issues/7162
closed 2020-02-27 by larsoner