mne-tools/mne-python #7239: error when trying to use pyvista to plot stc

SophieHerbst, 2020-01-22:
I am trying to use the new pyvista-plotting for stc (through spyder), 
on my own data, and this [mne-example](https://mne.tools/stable/auto_examples/visualization/plot_xhemi.html#sphx-glr-auto-examples-visualization-plot-xhemi-py)

When I do not specify a backend, it uses majavi. 
I tried: `mne.viz.set_3d_backend('pyvista')`

Output of `stc.plot`:
>   File "/home/sh254795/anaconda3/lib/python3.7/site-packages/mne/viz/_brain/_brain.py", line 201, in __init__
>     shape=(n_row, n_col), fig=figure)
> 
>   File "/home/sh254795/anaconda3/lib/python3.7/site-packages/mne/viz/backends/_pyvista.py", line 141, in __init__
>     self.plotter.disable_depth_peeling()
> 
> AttributeError: 'Plotter' object has no attribute 'disable_depth_peeling'

Output of ` python -c "import mne; mne.sys_info()"`:
> mne:           0.20.dev0
> numpy:         1.17.2 {blas=mkl_rt, lapack=mkl_rt}
> scipy:         1.3.1
> matplotlib:    3.1.1 {backend=Qt5Agg}
> 
> sklearn:       0.21.3
> numba:         0.45.1
> nibabel:       2.5.1
> cupy:          Not found
> pandas:        0.25.1
> dipy:          1.0.0
> mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.9.2}
> pyvista:       0.22.4
> vtk:           8.1.2

GuillaumeFavelier, 2020-01-22:
Can you try with the latest version [`0.23.0`](https://github.com/pyvista/pyvista/releases) of PyVista ?

Depth peeling have been introduced only recently.

SophieHerbst, 2020-01-22:
Ok thanks! 
No more error, but the plot appears in the spyder console and I cannot get it out, i.e. use interactive mode. All the usual setting preferences for ipython console graphics or `%matplotlib qt `does not help. Any ideas?

GuillaumeFavelier, 2020-01-22:
This is a false positive from the internal `check_notebook()` function. This is unfortunate.

GuillaumeFavelier, 2020-01-22:
I'll work on a patch right now :+1: . But basically, the plotter has a simple routine to detect a notebook environment and it seems that it didn't work properly in your case. The plotter should behave properly in a regular ipython console for some reason it's not the case in Spyder, thanks for reporting.

mmagnuski, 2020-01-29:
@GuillaumeFavelier Is pysurfer needed for plotting with pyvista? Because the last time I tried using mne-python stc plotting with pyvista backend in an env without pysrufer I got an error (the error said I should have pysurfer, but I'm not sure that should be the case). It works with pysurfer, but I thought it should work without.

larsoner, 2020-01-29:
We have some check for PySurfer still implemented that we should only use if `Mayavi` is the selected backend:
```
$ pip uninstall mayavi
$ python -c stc.py
Traceback (most recent call last):
  File "/home/larsoner/Desktop/stc.py", line 11, in <module>
    brain = stc.plot(subjects_dir=subjects_dir, initial_time=initial_time,
  File "/home/larsoner/python/mne-python/mne/source_estimate.py", line 1467, in plot
    brain = plot_source_estimates(
  File "</home/larsoner/python/mne-python/mne/externals/decorator.py:decorator-gen-128>", line 2, in plot_source_estimates
  File "/home/larsoner/python/mne-python/mne/utils/_logging.py", line 90, in wrapper
    return function(*args, **kwargs)
  File "/home/larsoner/python/mne-python/mne/viz/_3d.py", line 1687, in plot_source_estimates
    raise RuntimeError('This function requires pysurfer version '
RuntimeError: This function requires pysurfer version >= 0.9
```
@GuillaumeFavelier can you fix this so that it only checks in the case that mayavi is the selected backend?

PR #7266 (2020-01-30): MRG, FIX: Dont require PySurfer for PyVista
Closes #7239
Closes #7198

@mmagnuski feel free to try it

@GuillaumeFavelier this refactors some of the logic of how the `renderer` namespace is populated. This makes it so that we don't need to nest imports in the `backend` submodules anymore, and I think makes the logic/flow a bit clearer. WIP because I'm not 100% sure it's correct, feel free to look and let me know if it makes sense to you.
closed 2020-01-30 by larsoner