mne-tools/mne-python #7243: Brain Vision: Assertion error with channel names containing spaces

major-s, 2020-01-23:
Hi,
I have plenty of Brain Vision files with channel names containing spaces:
```
Channels
--------
#     Name      Phys. Chn.    Resolution / Unit   Low Cutoff [s]   High Cutoff [Hz]   Notch [Hz]    Series Res. [kOhm] Gradient         Offset
1     ECoG 1      1                 10 µV             DC              250              Off                0                 
2     ECoG 2      2                 10 µV             DC              250              Off                0      
```

and when I try to read them, I get following Assertion Error:
```
File "...\lib\site-packages\mne\io\brainvision\brainvision.py", line 637, in _get_vhdr_info
    assert ch in line_amp

AssertionError
```
because the channel name "ECoG 1" is splited into "ECoG" and "1" in the line 636
```line_amp = re.split(divider, settings[idx_amp + i])```
and then the channel name is not found in the resulting list.
It works, when I change the line 611 from
```divider = r'\s+'```
to 
```divider = r'\s\s+'```
and remove all the shift related code for the highpass and lowpass search but this will of course break in case of names with multiple white spaces or probably also very long channel names.

agramfort, 2020-01-23:
do you have time to send us a PR to address these issues?

major-s, 2020-01-24:
> 
> do you have time to send us a PR to address these issues?

I tried my best. Never used pull requests before. Unfortunately I had no time for writing any tests...

agramfort, 2020-01-24:
ok no problem. i'll give you a hand by pushing to this PR.

PR #7252 (2020-01-27): Fix problems with spaced names in brainvision.py #7243
Fixes broken reading of Brain Vision files with channel names containing spaces. Now the name is looked up in before split and the shift for reading highpass and lowpass filter settings is corrected. It works with multiple or no spaces.

closes #7243
closed 2020-01-27 by larsoner