mne-tools/mne-python #7244: Problem importing data into EEGLAB

amitkanthi, 2020-01-23:
I am facing a peculiar problem while importing .mat and .raw file into EEGLAB.  When I import data using .mat file then the EEGLAB doesn't import the events along with it. When I import data using .raw file then the EEGLAB doesn't show up the reference channel. Moreover, i failed to input the reference channel manually. 
Please help me with this issue.

larsoner, 2020-01-23:
@amitkanthi this sounds like an EEGLAB problem. I don't think we support writing `.mat` files, only `.fif` files. So perhaps this issue should be raised with EEGLAB support?

cbrnr, 2020-01-23:
Or did you mean MNELAB (which does support exporting to EEGLAB .set, which is basically just a .mat file)? If so, please report the issue with more details at https://github.com/cbrnr/mnelab.

amitkanthi, 2020-01-24:
hello

sorry. my bad. Actually its an EEGLAB problem. Extremely sorry for the inconvenience.

thanks. Appreciate your response.

closed 2020-01-24 by amitkanthi