mne-tools/mne-python #7248: 'CNTEventType1' object has no attribute 'Latency'

quocdat32461997, 2020-01-24:
Hi, 

I am trying to read raw EEG from .cnt files. However, Python3 returns the below error. The dataset I am reading is from UC San Diego https://sccn.ucsd.edu/~arno/fam2data/publicly_available_EEG_data.html. 

The dataset can be easily read by EEGLAB, however, I try to do everythig in Python3. 
If comment the lines shown in the Errors, I still receive the correct result. 

`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/site-packages/mne/io/cnt/cnt.py", line 169, in read_raw_cnt
    preload=preload, stim_channel=stim_channel, verbose=verbose)
  File "/usr/local/lib/python3.7/site-packages/mne/io/cnt/cnt.py", line 429, in __init__
    _read_annotations_cnt(input_fname, data_format=data_format))
  File "/usr/local/lib/python3.7/site-packages/mne/io/cnt/cnt.py", line 88, in _read_annotations_cnt
    duration = np.array([e.Latency for e in my_events], dtype=float)
  File "/usr/local/lib/python3.7/site-packages/mne/io/cnt/cnt.py", line 88, in <listcomp>
    duration = np.array([e.Latency for e in my_events], dtype=float)
AttributeError: 'CNTEventType1' object has no attribute 'Latency'`

agramfort, 2020-01-24:
which specific file did you try to read?

>

quocdat32461997, 2020-01-24:
> which specific file did you try to read?
> […](#)

They are .cnt files recorded by Neuroscan devices.

agramfort, 2020-01-25:
The page you point to contains many files. Can you help us by telling which one we should test first ? It saves time for us. Thanks

quocdat32461997, 2020-01-27:
I am sorry. 
Below is the link to my drive for one .cnt file. 
https://drive.google.com/file/d/1BNHjcJUSWLXSpXQ1aeVZ3KS4XzX-CfLS/view?usp=sharing

PR #7261 (2020-01-28): MRG fix for CNTEventType1 that don't have Latency
closes #7248

@quocdat32461997 can you have a look and test? I tried comparing the data with EEGLAB but I get here 101 events while EEGLAB gives me 100 (it skips the last one).
closed 2020-01-28 by larsoner