mne-tools/mne-python #7254: BUG: possible bug wrt BallTree xhs update

zuxfoucault, 2020-01-26:
#### Describe the bug
The [xhs](https://github.com/mne-tools/mne-python/blob/master/mne/surface.py#L498) didn't get updated after first initialized as [[[0. 0. 0.]]](https://github.com/mne-tools/mne-python/blob/master/mne/gui/_coreg_gui.py#L487). Therefore, the [nearest_transformed_high_res_mri_idx_hsp](https://github.com/mne-tools/mne-python/blob/master/mne/gui/_coreg_gui.py#L458) is always 0.
[image]

After the [snippet](https://github.com/zuxfoucault/mne-python/blob/fix_balltree_xhs_update/mne/gui/_coreg_gui.py#L495) added, the `xhs` is updated with `transformed_high_res_mri_points`, and the `icp_fid_match = nearest`  performed better than `matched` (median distance: 1.280 mm vs 1.642 mm).
[image]

#### Steps to reproduce
https://14813-1301584-gh.circle-artifacts.com/0/dev/auto_examples/forward/plot_auto_alignment.html
Note: the example isn't really ready for public consumption as far as I know

larsoner, 2020-01-27:
@zuxfoucault did you need to change code in [_coreg_gui.py](https://github.com/zuxfoucault/mne-python/blob/fix_balltree_xhs_update/mne/gui/_coreg_gui.py#L495) to fix things, or just add some lines to the example? If you just needed to change the example code, then I think we can close since that example was never merged / never fully tested / is not yet ready for public consumption.

zuxfoucault, 2020-01-27:
@larsoner Just add the notification part in [_coreg_gui.py](https://github.com/zuxfoucault/mne-python/blob/fix_balltree_xhs_update/mne/gui/_coreg_gui.py#L495)

larsoner, 2020-01-27:
Something is weird about https://github.com/zuxfoucault/mne-python/commit/698d3240553e9cad580d9014e60440d57d46a833 because the [existing _update_nearest_calc](https://github.com/mne-tools/mne-python/blob/master/mne/gui/_coreg_gui.py#L490-L493) should already take care of it:
```
    @on_trait_change('processed_high_res_mri_points')
    def _update_nearest_calc(self):
        self.nearest_calc = _DistanceQuery(
            self.processed_high_res_mri_points * self.parameters[6:9])
```

larsoner, 2020-01-27:
Also, we (should) always do the [nearest calculation in MRI space](https://github.com/mne-tools/mne-python/pull/6732), so your change will break things if display is in head space because it's using the transformed points

zuxfoucault, 2020-01-27:
> Something is weird about [zuxfoucault@698d324](https://github.com/zuxfoucault/mne-python/commit/698d3240553e9cad580d9014e60440d57d46a833) because the [existing _update_nearest_calc](https://github.com/mne-tools/mne-python/blob/master/mne/gui/_coreg_gui.py#L490-L493) should already take care of it:
> 
> ```
>     @on_trait_change('processed_high_res_mri_points')
>     def _update_nearest_calc(self):
>         self.nearest_calc = _DistanceQuery(
>             self.processed_high_res_mri_points * self.parameters[6:9])
> ```

My rationale is that (self.processed_high_res_mri_points)[https://github.com/zuxfoucault/mne-python/blob/fix_balltree_xhs_update/mne/gui/_coreg_gui.py#L682] is generated after
(self.nearest_transformed_high_res_mri_idx_hsp)[https://github.com/zuxfoucault/mne-python/blob/fix_balltree_xhs_update/mne/gui/_coreg_gui.py#L683], where xhp is still initialized as [[[0. 0. 0.]]](https://github.com/mne-tools/mne-python/blob/master/mne/gui/_coreg_gui.py#L487).

larsoner, 2020-01-27:
> My rationale is that self.processed_high_res_mri_points is generated after self.nearest_transformed_high_res_mri_idx_hsp

Where does this happen? I see the def as:
```
    processed_high_res_mri_points = Property(
        depends_on=['mri:bem_high_res:surf', 'grow_hair'])
```
So it does not depend on `nearest_transformed_high_res_mri_idx_hsp`

zuxfoucault, 2020-01-27:
> Where does this happen? I see the def as:
>
> ```
>     processed_high_res_mri_points = Property(
>         depends_on=['mri:bem_high_res:surf', 'grow_hair'])
> ```
>
> So it does not depend on [...]

I think it's [here](https://github.com/zuxfoucault/mne-python/blob/fix_balltree_xhs_update/mne/gui/_coreg_gui.py#L682-L683) 
`nearest_transformed_high_res_mri_idx_hsp` is index for `processed_high_res_mri_points`

larsoner, 2020-01-27:
It's used as an index but I don't see why that should affect `nearest_calc`. The code you linked to is inside `_setup_icp`, which should only be called once `processed_high_res_mri_points` is already populated properly, `nearest_calc` is set up properly (automatically), and you've asked to do ICP.

I think the way to solve this is to figure out exactly how/when these things are being triggered, i.e. the flow of updates/changes. It's possible that we are missing a trigger somewhere, but the fix in this PR does not seem right, even if it happens to make things work.

zuxfoucault, 2020-01-28:
History of some function call printed:
[image]

It seems when `_get_processed_high_res_mri_points` is first initialized, handler `_update_nearest_calc: processed_high_res_mri_points` wouldn't be called. The same applies to `_get_transformed_high_res_mri_points`; handler `_update_nearest_calc: transformed_high_res_mri_points` 
 is called on the second `_get_transformed_high_res_mri_points` call. Howeve, I still haven't figured out why `print('_get_processed_high_res_mri_points')` inside `def _get_processed_high_res_mri_points' dosen't get called after first initilization.

But it looks the initialization of `processed_high_res_mri_points` doesn't get nearest_calc set up with `_DistanceQuery(self.processed_high_res_mri_points * self.parameters[6:9])`

zuxfoucault, 2020-01-28:
> Something is weird about [zuxfoucault@698d324](https://github.com/zuxfoucault/mne-python/commit/698d3240553e9cad580d9014e60440d57d46a833) because the [existing [...]

Out of curiosity, as a workaround, would this look better? `self.nearest_calc = _DistanceQuery(
            self.transformed_high_res_mri_points)` is the original proposal is replaced with `self.nearest_calc = _DistanceQuery( self.processed_high_res_mri_points * self.parameters[6:9])`, hence
```
    @on_trait_change('transformed_high_res_mri_points')
    def _update_nearest_calc(self):
        self.nearest_calc = _DistanceQuery(
             self.processed_high_res_mri_points * self.parameters[6:9])
```

zuxfoucault, 2020-01-28:
Just adding a [dependence](https://github.com/zuxfoucault/mne-python/blob/fix_balltree_xhs_update/mne/gui/_coreg_gui.py#L243) on `mri_trans` for `processed_high_res_mri_points` would evoke the update, although it's surplus after the 2nd call on `_get_processed_high_res_mri_points`
[image]

PR #7272 (2020-02-05): MRG, FIX: Fix bug with update
Closes #7254

@zuxfoucault can you confirm this fixes your problem?

The fix I came up with with was just to set `_nearest_calc_default` to actually use the processed points. The `ArrayOrNone` changes are just to help us ensure that we don't use values before we should, and thankfully they do not break anything at least in local testing.
closed 2020-02-01 by zuxfoucault