mne-tools/mne-python #7255: reading channel locations from .set files

FedericaDegno, 2020-01-26:
I am working with EEG epoched data saved as .set files (and the corresponding .fdt files from EEGlab). The data were recorded from 66 channels (64 EEG and 2 EOG) from EasyCaps, with 10-20 IS. In Matlab, the EEG.chanlocs correctly shows the coordinates of each electrode (labels, type, theta, radius, X, Y, Z, sph_theta, sph_phi, sph_radius, urchin, ref). But it seems that I cannot read these locations in MNE Python.

This is the code I used:

import mne
data_path = r"D:\EEGdata";
fname = data_path + '\ppt10.set'
mydata = mne.io.read_epochs_eeglab(fname)
mydata
mydata.plot()
mydata.ch_names

The .set files are imported ok, and channel labels are correctly displayed, but the channel locations are not found.
mydata.plot_sensors()     #RuntimeError: No valid channel positions found

Any suggestion?

larsoner, 2020-01-27:
What are:
```
'eeg' in mydata
mydata.ch_names[0]
mydata.info['chs'][0]['loc']
mydata.info['chs'][0]['coil_type']
```
Or if you can upload the data somewhere (e.g., DropBox) someone can take a look.

dede3de, 2020-01-27:
```
'eeg' in mydata
Out[45]: True

mydata.ch_names[0]
Out[46]: 'FP1'

mydata.info['chs'][0]['loc']
Out[47]: array([nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan])

mydata.info['chs'][0]['coil_type']
Out[48]: 1 (FIFFV_COIL_EEG)
```

larsoner, 2020-01-27:
Okay then something is weird about how we are dealing with `EEG.chanlocs` it looks like. Can you upload the files somewhere?

dede3de, 2020-01-27:
I have uploaded an example dataset (.set and .fdt) here: https://drive.google.com/drive/folders/1kT392TAWm-0DWGEbj00b6EGgRqwbNifX?usp=sharing

mmagnuski, 2020-01-28:
@dede3de Are you sure you've placed the files there? When I click on the link I get an empty directory.

agramfort, 2020-01-28:
indeed it's empty

>

dede3de, 2020-01-28:
Try this one: https://drive.google.com/open?id=1kT392TAWm-0DWGEbj00b6EGgRqwbNifX

mmagnuski, 2020-01-28:
@dede3de Sorry, it's also empty. :(

agramfort, 2020-01-28:
I confirm

dede3de, 2020-01-28:
Ok, not sure what went wrong with google drive. Here is Dropbox: 
https://www.dropbox.com/s/0ug9re25ijv88p0/ppt10_toFT.set?dl=0
https://www.dropbox.com/s/5144xgs71mnrlqm/ppt10_toFT.fdt?dl=0
I hope it now works.

agramfort, 2020-01-29:
it's indeed a bug that should be fixed in #7262

thanks for the bug report.

closed 2020-01-29 by larsoner

parasmemon, 2020-06-07:
I am also getting the same error: mydata.plot_sensors() #RuntimeError: No valid channel positions found

agramfort, 2020-06-07:
then you did not set a proper montage for your data or you have an issue of
channel types

>

parasmemon, 2020-06-07:
Hello Agramfort,

I think I have issue of channel position, my data is based on standard 10-20 EEG location system
Please check code and error.

Code:
ten_twenty_montage = mne.channels.make_standard_montage('standard_1020')
print(ten_twenty_montage)

========
fig = plt.figure()
ax2d = fig.add_subplot(121)
ax3d = fig.add_subplot(122, projection='3d')
raw.plot_sensors(ch_type='eeg', axes=ax2d)
raw.plot_sensors(ch_type='eeg', axes=ax3d, kind='3d')
ax3d.view_init(azim=70, elev=15)

RuntimeError: No valid channel positions found

agramfort, 2020-06-07:
where do you do:

raw.set_montage(ten_twenty_montage)

?

do you see then any warning or error?

>

parasmemon, 2020-06-07:
Yup, I did this and got this error:

ValueError: DigMontage is a only a subset of info. There are 40 channel positions not present it the DigMontage. The required channels are: {'14', '2', '16', '4', '13', '19', '21', '17', '33', '5', '15', '9', '26', '10', '36', '38', '31', '11', '29', '1', '24', '30', '37', '12', '27', '22', '34', '20', '7', '35', '32', '6', '18', '25', '28', '8', '40', '3', '23', '39'}.
You can use `raise_if_subset=False` in `set_montage` to avoid this ValueError and get a DeprecationWarning instead.

parasmemon, 2020-06-07:
This is the python code:

Code:

'eeg' in raw
True
raw.ch_names[0]
'1'
raw.info['chs'][0]['loc']
array([nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan])
raw.info['chs'][0]['coil_type']
1 (FIFFV_COIL_EEG)

raw.plot_sensors(show=True)
or
raw.plot_sensors(ch_type='eeg')

This give me error: RuntimeError: No valid channel positions found

agramfort, 2020-06-07:
then set_montage did not work

I suppose your channels are  {'14', '2', '16', '4', '13', '19', '21',
'17', '33', '5', '15', '9', '26', '10', '36', '38', '31', '11', '29',
'1', '24', '30', '37', '12', '27', '22', '34', '20', '7', '35', '32',
'6', '18', '25', '28', '8', '40', '3', '23', '39'}. but 10-20 system
does not use this naming.

You need to fix your channel names before setting the montage

parasmemon, 2020-06-07:
I have found that montage has these channels name:
montage = mne.channels.read_montage('standard_1020')
print(montage.ch_names)
['LPA', 'RPA', 'Nz', 'Fp1', 'Fpz', 'Fp2', 'AF9', 'AF7', 'AF5', 'AF3', 'AF1', 'AFz', 'AF2', 'AF4', 'AF6', 'AF8', 'AF10', 'F9', 'F7', 'F5', 'F3', 'F1', 'Fz', 'F2', 'F4', 'F6', 'F8', 'F10', 'FT9', 'FT7', 'FC5', 'FC3', 'FC1', 'FCz', 'FC2', 'FC4', 'FC6', 'FT8', 'FT10', 'T9', 'T7', 'C5', 'C3', 'C1', 'Cz', 'C2', 'C4', 'C6', 'T8', 'T10', 'TP9', 'TP7', 'CP5', 'CP3', 'CP1', 'CPz', 'CP2', 'CP4', 'CP6', 'TP8', 'TP10', 'P9', 'P7', 'P5', 'P3', 'P1', 'Pz', 'P2', 'P4', 'P6', 'P8', 'P10', 'PO9', 'PO7', 'PO5', 'PO3', 'PO1', 'POz', 'PO2', 'PO4', 'PO6', 'PO8', 'PO10', 'O1', 'Oz', 'O2', 'O9', 'Iz', 'O10', 'T3', 'T5', 'T4', 'T6', 'M1', 'M2', 'A1', 'A2']... These are 97 channels.

In my data channels names are [1,2,3,...,40]

How can I rename my 40 channels according to the '10-20' montage?

larsoner, 2020-06-08:
> How can I rename my 40 channels according to the '10-20' montage?

This depends on the details of your recording setup, you'll have to look into how the data were recorded to know this

parasmemon, 2020-06-08:
Description of my data: "According to the standard 10-20 EEG location system, on both right and left hemispheres, eight four-millimeter electrodes were placed in an array along important language production (Broca’s area; F3 and F4) and language comprehension (Wernicke’s area; P5 and P6) regions of the human brain 9 (N=32)."

Renaming: mne.rename_channels(raw, ('1', 'LPA'))

Error: AttributeError: 'RawBrainVision' object has no attribute '_check_consistency'

Also, I can not understand which channel name is replaced with the montage channel name.

larsoner, 2020-06-08:
> Renaming: mne.rename_channels(raw, ('1', 'LPA'))

The docstring states in requires an Info object, you passed it a Raw instance. You want `rename_channels(raw.info, ...)` or just `raw.rename_channels`. The second arg also needs to be a dict, you passed a tuple.

> Also, I can not understand which channel name is replaced with the montage channel name.

It's also not clear to me from your description, you'll have to look into it and discuss with the people you got the data from

parasmemon, 2020-06-08:
If I am using this python code:
mne.rename.channels(raw.info, {'1': 'LPA'}) // 1 is my data channel name and LPA is montage channel name.

I am getting an error:  
AttributeError: module 'mne' has no attribute 'rename'

agramfort, 2020-06-08:
rename_channels not rename.channels ....

>

parasmemon, 2020-06-08:
Ok thanks.
mne.rename_channels(raw.info, {'1': 'LPA'})

Again getting an error:
ValueError: Channel name(s) in mapping missing from info: ['1']

parasmemon, 2020-06-09:
@agramfort @larsoner 
I have put my description data file in the drive folder. Can you pls look at it?
I am really stuck in this channel names and sensor location issue.

https://drive.google.com/drive/u/0/folders/1lLCpfeHON_ihIk4ZKIDJuujWDBzMigMihttps://drive.google.com/drive/u/0/folders/1lLCpfeHON_ihIk4ZKIDJuujWDBzMigMi

drammock, 2020-06-09:
@parasmemon Please don't post any more questions to this thread; this issue has been closed for several months and is not the right place to ask usage questions. For further help, please use the [MNE mailing list](http://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis) or the [gitter channel](https://gitter.im/mne-tools/mne-python) if you need additional help.

PR #7262 (2020-01-29): cleanup deprecation of montage setting in read_epochs_eeglab
closes #7255