mne-tools/mne-python #7260: Minor: epoch counts in epochs.plot() starts with 1 instead of 0

SophieHerbst, 2020-01-28:
When using `epochs.plot()` to display epochs, e.g. to manually identify bad ones, the epoch numbers displayed on the plot start with 1, not 0, `while epochs.drop()` expects counts to start from 0. 
[I am aware that one can use the interactive plot + annotations to manually reject epochs, but still think it would avoid confusion if the counters match.]

agramfort, 2020-01-28:
we've had this issue in the past for ICA component numbering.
I think we ended up using 0 based-labels for plots using the # sign.
To be checked...

SophieHerbst, 2020-01-28:
Yes, at least in my output ICA components are numbered form 0. Can the same be done for the epochs?

larsoner, 2020-01-28:
Sounds good to me, @SophieHerbst can you make a PR?

SophieHerbst, 2020-01-28:
I can give it a try in the next couple of days.

PR #7299 (2020-02-11): MRG, FIX: make labels in epochs.plot start at 0, fixes issue #7260
Makes epochs.plot() start at 0 instead of 1 (issue #7260). 
Note: a bug with spyder (#6528) prevents me from testing what happens after closing the interactive plot. I checked that the indices returned for manually selected epochs start with 0, too. 

Closes #7260
closed 2020-02-11 by larsoner