mne-tools/mne-python #7268: pyvista stc plots: move through time constantly by holding key

SophieHerbst, 2020-01-30:
@GuillaumeFavelier maybe this already exists: is there a way to move continuously through time on an stc, by holding for instance the left/right arrow key?  The playback speed scaler sets the step-size on the time-axis, but I still need to move the mouse across the time scale, which is a bit inconvenient, and hard to do continuously.

agramfort, 2020-01-30:
did you try pressing space bar?

>

SophieHerbst, 2020-01-30:
I did not (what a narrow concept of time). Cool!
What is the shortcut to bring up a legend / help ?

agramfort, 2020-01-30:
>
> What is the shortcut to bring up a legend / help ?
>

it does not exist yet :) to be consistent with other plots it should be 'h'

GuillaumeFavelier, 2020-01-30:
That would be nice to have considering the number of toggles/shortcuts has increased in TimeViewer. I added this idead in #7162 :+1:

larsoner, 2020-01-30:
This window that pops up can just be a new / separate `matplotlib` figure like we do in all of our other plotters. @GuillaumeFavelier it might be worth refactoring the code to make a new function that takes a list of `(shortcut, description)` pairs and pops up a new window with the list.

GuillaumeFavelier, 2020-01-31:
If we plan to reuse `MplCanvas`, this requires https://github.com/mne-tools/mne-python/pull/7247 to be merged

larsoner, 2020-01-31:
I wouldn't reuse `MplCanvas`, I would actually pop up a window for this one. So you can just do a new `plt.figure`

closed 2020-02-13 by larsoner