mne-tools/mne-python #7282: error when saving TFR: unsupported type <class 'datetime.datetime'>

SophieHerbst, 2020-02-04:
I am computing power on epochs and want to save the output. Since I updated MNE to 
0.20.dev0, I get the following error: 

```
  File "/home/sh254795/anaconda3/lib/python3.7/site-packages/mne/externals/h5io/_h5io.py", line 225, in _triage_write
    raise TypeError(err_str)

TypeError: unsupported type <class 'datetime.datetime'> (in <class 'list'>[0][1]["info"]["meas_date"])

```
Here is an example that reproduces the error:

```
import mne
from mne.datasets import sample

print(__doc__)

###############################################################################
# Set parameters
# --------------
data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_raw.fif'
tmin, tmax, event_id = -0.3, 0.6, 1

# Setup for reading the raw data
raw = mne.io.read_raw_fif(raw_fname)
events = mne.find_events(raw, stim_channel='STI 014')

include = []
raw.info['bads'] += ['MEG 2443', 'EEG 053']  # bads + 2 more

# picks MEG gradiometers
picks = mne.pick_types(raw.info, meg='grad', eeg=False, eog=True,
                       stim=False, include=include, exclude='bads')

# Load condition 1
event_id = 1
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, picks=picks,
                    baseline=(None, 0), preload=True,
                    reject=dict(grad=4000e-13, eog=150e-6))

# Take only one channel
ch_name = 'MEG 1332'
epochs.pick_channels([ch_name])

decim = 5
freqs = np.arange(8, 40, 2)  # define frequencies of interest
sfreq = raw.info['sfreq']  # sampling in Hz

power, itc = mne.time_frequency.tfr_morlet(epochs, freqs=freqs, return_itc=True, n_cycles=4., decim=decim)

power.save('TMP', overwrite=True)
```

agramfort, 2020-02-04:
I can replicate. let me have a look

agramfort, 2020-02-04:
in the mean time you can do:

del power.info['meas_date']

before saving.

agramfort, 2020-02-05:
Fixed in #7284

closed 2020-02-05 by agramfort