mne-tools/mne-python #7283: plotting: address mpl deprecation warning when calling plt.subplots() several times.

sappelhoff, 2020-02-04:
I am plotting several topographies using `mne.viz.plot_evoked_topomap` so that for different conditions I can get a single figure with:

- each row is a condition
- each column is a different time point
- each panel is a topography

this is my code:

```Python
def plot_topos(erp_dict):
    """Plot a topography over time."""
    # Figure settings
    step = 0.1
    times = np.arange(-0.1, 0.8, step)
    average = step
    
    # Set up figure
    n_keys = len(erp_dict)
    fig, ax = plt.subplots(n_keys, len(times), figsize=(15, 2*n_keys))
    ax = np.atleast_2d(ax)
    
    for ii, key in enumerate(erp_dict):
        # Form grand average
        ga = mne.grand_average(erp_dict[key])        
        mima = np.abs(ga.data).max() * 1e6
        mne.viz.plot_evoked_topomap(ga, times=times, vmin=-mima, vmax=mima,
                                    axes=ax[ii, :], colorbar=True)
        ax[ii, 0].set_ylabel(key)
        
        
    fig.suptitle('Sample ERP')

    return fig
```

Using this, I run into the following warning:

> MatplotlibDeprecationWarning: Adding an axes using the same arguments as a previous axes currently reuses the earlier instance.  In a future version, a new instance will always be created and returned.  Meanwhile, this warning can be suppressed, and the future behavior ensured, by passing a unique label to each axes instance.

I think this is triggered by the following line:

https://github.com/mne-tools/mne-python/blob/f474eb196eca50104f09948c6a4215d628a971f5/mne/viz/topomap.py#L1682

We can probably fix it by changing the line as follows:

```diff
-cax = plt.subplot(1, n_fig_axes + 1, n_fig_axes + 1)
+cax = plt.gcf().add_subplot(1, n_fig_axes + 1, n_fig_axes + 1)
```

Has anyone else encountered this issue? Is this a sensible solution? Should I make a PR?

larsoner, 2020-02-04:
Can you fix it by only doing `colorbar=True` on the first or last iteration, and `colorbar=False` otherwise?

sappelhoff, 2020-02-05:
yes, that fixes it. Thanks @larsoner

PR #7284 (2020-02-05): MRG, FIX: HDF5 datetime bug
closes #7283

@larsoner any suggested way to fix this?
closed 2020-02-05 by sappelhoff