mne-tools/mne-python #7287: ";" are not supported in EDF annotation descriptions

adam2392, 2020-02-04:
#### Describe the bug
Hi, I'm currently reading in an EDF file w/ the latest MNE dev version. read_raw_edf, and I encounter this error:

"ValueError: Semicolons in descriptions not supported."

I took a look at the descriptions for each annotation inside the EDF file using edfbrowser, and I see semicolons, but at the same time I don't quite understand why this is a problem? Is it possible to submit a PR that gets rid of this check if it is not necessary?

I can submit a PR if this is something we can fix. 

#### Steps to reproduce
1. Create EDF file with annotation with a ";" in description.
2. Use read_raw_edf to read in file

#### Expected results
Some specific error text, specifying how we can fix this? 

Or just no error at all.

#### Actual results
"ValueError: Semicolons in descriptions not supported."

#### Additional information
〈Replace this text with the output of `mne.sys_info()`.〉

agramfort, 2020-02-04:
Sharing a file or fixing the bug directly is helpful

adam2392, 2020-02-04:
> Sharing a file or fixing the bug directly is helpful

Will try trimming a file and posting a dropbox link here.

adam2392, 2020-02-04:
https://www.dropbox.com/s/f3cg7xme8x3a0ui/SEEG_SZ_1P_0001_reduced_reduced.edf?dl=0

for a reduced file that will not read because ";" are in annotation descriptions. I took a look at the source code, and:

```
def _check_o_d_s(onset, duration, description):
    ...
    description = np.array(description, dtype=str)
    if description.ndim == 0 or description.shape == (1,):
        description = np.repeat(description, len(onset))
    if description.ndim != 1:
        raise ValueError('Description must be a one dimensional array, '
                         'got %d.' % (description.ndim,))
    if any([';' in desc for desc in description]):
        raise ValueError('Semicolons in descriptions not supported.')
```
is causing the error. I don't see why, we possibly can't take out that value error, or do a string replacement w/ say "-", or "|" to denote separation if really necessary?

PR #7288 (2020-02-11): [MRG] Annotation fix to allow Semicolons in Description
Thanks for contributing a pull request! Please make sure you have read the
[contribution guidelines](https://mne.tools/dev/install/contributing.html)
before submitting.

#### Reference issue
Fixes: #7287 

#### What does this implement/fix?
WIP, attempts to check if removing the ";" check breaks anything.

#### Additional information
See Issue for sample data.
closed 2020-02-11 by larsoner