mne-tools/mne-python #7300: Error in mne.io.read_raw_curry

mabhijithn, 2020-02-08:
**READ THIS FIRST:** If you are having trouble getting MNE-Python to work with
your own data, you should probably ask for help on our
[Gitter channel](https://gitter.im/mne-tools/mne-python) or on our
[mailing list](https://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis)
before opening an issue; you will get help faster that way. See
[the FAQ entry on filing bug reports](https://mne.tools/dev/overview/faq.html#i-think-i-found-a-bug-what-do-i-do)
for more guidance. If you're pretty sure your problem is a bug in MNE-Python,
**delete this section** and fill in the headings below, replacing the
placeholder text with your own responses. Surround any code samples with triple
backticks above and below the code block, or put code in a
[public gist](https://gist.github.com) and link to it in the issue.

#### Describe the bug
Error when reading curry file. The error is pasted in actual results.

#### Steps to reproduce

Test data is large in size. I am placing the download link here:
https://mega.nz/#F!pdMmxayI!eKQaoIN_VYMwCYHRhDS_tw
All curry files with .ceo, .dap, .dat and .rs3 extensions of the same name are shared in the above link

`fullfile = 'S23_AAD_2R'`
`eegdata = mne.io.read_raw_curry(fullfile)`

#### Expected results
The raw data should be available in **eegdata**

#### Actual results

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
~/Documents/python/example_scripts/test_aad.py in 
      2 file_name = 'S23_AAD_1L'
      3 fullfile = data_path+file_name
----> 4 S23_1L_raw = mne.io.read_raw_curry(fullfile)

 in read_raw_curry(fname, preload, verbose)

~/anaconda3/envs/eegsp/lib/python3.7/site-packages/mne/utils/_logging.py in wrapper(*args, **kwargs)
     88             with use_log_level(verbose_level):
     89                 return function(*args, **kwargs)
---> 90         return function(*args, **kwargs)
     91     return FunctionMaker.create(
     92         function, 'return decfunc(%(signature)s)',

~/anaconda3/envs/eegsp/lib/python3.7/site-packages/mne/io/curry/curry.py in read_raw_curry(fname, preload, verbose)
    272 
    273     """
--> 274     return RawCurry(fname, preload, verbose)
    275 
    276 

 in __init__(self, fname, preload, verbose)

~/anaconda3/envs/eegsp/lib/python3.7/site-packages/mne/utils/_logging.py in wrapper(*args, **kwargs)
     88             with use_log_level(verbose_level):
     89                 return function(*args, **kwargs)
---> 90         return function(*args, **kwargs)
     91     return FunctionMaker.create(
     92         function, 'return decfunc(%(signature)s)',

~/anaconda3/envs/eegsp/lib/python3.7/site-packages/mne/io/curry/curry.py in __init__(self, fname, preload, verbose)
    297         curry_paths = _get_curry_file_structure(fname, required=["info",
    298                                                                  "data",
--> 299                                                                  "labels"])
    300 
    301         data_fname = op.abspath(curry_paths.data)

~/anaconda3/envs/eegsp/lib/python3.7/site-packages/mne/io/curry/curry.py in _get_curry_file_structure(fname, required)
     49 
     50     # we don't use os.path.splitext to also handle extensions like .cdt.dpa
---> 51     fname_base, ext = fname.split(".", maxsplit=1)
     52     version = _get_curry_version(ext)
     53 

ValueError: not enough values to unpack (expected 2, got 1)

#### Additional information

import mne...
Platform:      Linux-4.15.0-74-generic-x86_64-with-debian-buster-sid
Python:        3.7.6 (default, Jan  8 2020, 19:59:22)  [GCC 7.3.0]
Executable:    /home/abhijith/anaconda3/envs/eegsp/bin/python
CPU:           x86_64: 8 cores
Memory:        15.5 GB

mne:           0.19.2
numpy:         1.18.1 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.4.1
matplotlib:    3.1.3 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.22.1
numba:         0.48.0
nibabel:       3.0.1
cupy:          Not found
pandas:        1.0.0
dipy:          1.1.1
mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.9.2}
pyvista:       0.23.1
vtk:           8.1.2

larsoner, 2020-02-10:
You need to pass an actual filename. In #7297 this will now do:
```
$ python -c "import mne; mne.io.read_raw_curry('S23_AAD_1L')"
Traceback (most recent call last):
...
  File "/home/larsoner/python/mne-python/mne/utils/check.py", line 148, in _check_fname
    raise IOError('%s "%s" does not exist' % (name, fname))
OSError: File "S23_AAD_1L" does not exist
```
To give you a hint that you should do:
```
$ python -c "import mne; mne.io.read_raw_curry('S23_AAD_1L.dat')"
```
There is currently a bug with this as the event file `.cef` does not exist. Is it supposed to be there? If not, we can probably allow reading with no events.

mabhijithn, 2020-02-10:
In our experiments we did not have an event file. The matlab function reads data even without this file.

mabhijithn, 2020-02-10:
Thanks for the tip!
I had tried to read with '.dat' extension. But the error
`Event file not found. No Annotations set.`  led me to input without extension. Can I edit the python functions myself to to read without `.cef` file?

larsoner, 2020-02-10:
@mabhijithn on the branch from https://github.com/mne-tools/mne-python/pull/7297 I cannot replicate your problem:
```
larsoner@bunk:~/Desktop/Raw-data$ python -c "import mne; print(mne.io.read_raw_curry('S23_AAD_2R.dat'))"
<RawCurry  |  S23_AAD_2R.dat, n_channels x n_times : 256 x 431400 (431.4 sec), ~478 kB, data not loaded>
```
But there does appear to be a different problem with your file:
```
larsoner@bunk:~/Desktop/Raw-data$ python -c "import mne; print(mne.io.read_raw_curry('S23_AAD_2R.dat', preload=True))"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "</home/larsoner/python/mne-python/mne/externals/decorator.py:decorator-gen-201>", line 2, in read_raw_curry
  File "/home/larsoner/python/mne-python/mne/utils/_logging.py", line 90, in wrapper
    return function(*args, **kwargs)
  File "/home/larsoner/python/mne-python/mne/io/curry/curry.py", line 394, in read_raw_curry
    return RawCurry(fname, preload, verbose)
  File "</home/larsoner/python/mne-python/mne/externals/decorator.py:decorator-gen-202>", line 2, in __init__
  File "/home/larsoner/python/mne-python/mne/utils/_logging.py", line 90, in wrapper
    return function(*args, **kwargs)
  File "/home/larsoner/python/mne-python/mne/io/curry/curry.py", line 427, in __init__
    super(RawCurry, self).__init__(
  File "</home/larsoner/python/mne-python/mne/externals/decorator.py:decorator-gen-144>", line 2, in __init__
  File "/home/larsoner/python/mne-python/mne/utils/_logging.py", line 90, in wrapper
    return function(*args, **kwargs)
  File "/home/larsoner/python/mne-python/mne/io/base.py", line 426, in __init__
    self._preload_data(preload)
  File "</home/larsoner/python/mne-python/mne/externals/decorator.py:decorator-gen-147>", line 2, in _preload_data
  File "/home/larsoner/python/mne-python/mne/utils/_logging.py", line 90, in wrapper
    return function(*args, **kwargs)
  File "/home/larsoner/python/mne-python/mne/io/base.py", line 671, in _preload_data
    self._data = self._read_segment(data_buffer=data_buffer)
  File "/home/larsoner/python/mne-python/mne/io/base.py", line 568, in _read_segment
    self._read_segment_file(data[:, this_sl], idx, fi,
  File "/home/larsoner/python/mne-python/mne/io/curry/curry.py", line 461, in _read_segment_file
    _read_segments_file(self, data, idx, fi, start, stop, cals,
  File "/home/larsoner/python/mne-python/mne/io/utils.py", line 220, in _read_segments_file
    raise RuntimeError('Incorrect number of samples (%s != %s), '
RuntimeError: Incorrect number of samples (19140608 != 24999936), please report this error to MNE-Python developers
```
Is it possible that the data was truncated during upload?

mabhijithn, 2020-02-11:
No.. I don't think so. I read the same data using the MATLAB function, and it was working fine.

PR #7297 (2020-02-13): MRG, FIX: Fix curry reader
Our curry reader currently does not populate `info['chs'][ii]['loc']`, `info['dig']`, or `info['dev_head_t']` properly. 

@DiGyt did you try to do any source localization, `plot_sensors`, etc. with the curry reader?

1000 nAm current phantom results have 2.1 mm error, 99%+ GOF:

[image]

Todo:

- [x] Add or mock some test data where `dev_head_t` is actually calculated
- [x] ~~Update `get_phantom_dipoles`~~
- [x] ~~Add phantom data as OSF.io dataset~~
- [x] ~~Add Curry phantom example~~

Closes #7300
closed 2020-02-13 by larsoner