mne-tools/mne-python #7302: No module named 'mne.datasets.megsim'

Ziaeemehr, 2020-02-09:
I get this error :

```python
from mne.datasets.megsim import load_data
ModuleNotFoundError: No module named 'mne.datasets.megsim'
```
```sh
mne-python python3 -c "import mne; mne.sys_info()"
Platform:      Linux-4.15.0-74-generic-x86_64-with-Ubuntu-18.04-bionic
Python:        3.6.9 (default, Nov  7 2019, 10:44:02)  [GCC 8.3.0]
Executable:    /usr/bin/python3
CPU:           x86_64: 8 cores
Memory:        7.6 GB

mne:           0.19.2
numpy:         1.18.1 {blas=openblas, lapack=openblas}
scipy:         1.4.1
matplotlib:    3.1.2 {backend=Qt5Agg}

sklearn:       0.21.3
numba:         0.45.1
nibabel:       2.5.0
cupy:          6.3.0
pandas:        0.25.3
dipy:          1.0.0
mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.14.0}
pyvista:       0.23.1
vtk:           8.1.2
```

I used this for installation:

```sh
sudo -H pip3 install mne
```

Ziaeemehr, 2020-02-09:
I also installed the mne-python from source file of github repo. 
I get the same error.

I also have this error:
```python
from mne.datasets import sample
data_path = sample.data_path()
```

```sh
Using default location ~/mne_data for sample...
Downloading archive MNE-sample-data-processed.tar.gz to /home/abolfazl/mne_data
Error while fetching file https://osf.io/86qa2/download?version=5. Dataset fetching aborted.
Traceback (most recent call last):
  File "/usr/lib/python3.6/urllib/request.py", line 1318, in do_open
    encode_chunked=req.has_header('Transfer-encoding'))
  File "/usr/lib/python3.6/http/client.py", line 1254, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.6/http/client.py", line 1300, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.6/http/client.py", line 1249, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.6/http/client.py", line 1036, in _send_output
    self.send(msg)
  File "/usr/lib/python3.6/http/client.py", line 974, in send
    self.connect()
  File "/usr/lib/python3.6/http/client.py", line 1415, in connect
    server_hostname=server_hostname)
  File "/usr/lib/python3.6/ssl.py", line 407, in wrap_socket
    _context=self, _session=session)
  File "/usr/lib/python3.6/ssl.py", line 817, in __init__
    self.do_handshake()
  File "/usr/lib/python3.6/ssl.py", line 1077, in do_handshake
    self._sslobj.do_handshake()
  File "/usr/lib/python3.6/ssl.py", line 689, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/abolfazl/.vscode/extensions/ms-python.python-2020.1.58038/pythonFiles/lib/python/new_ptvsd/no_wheels/ptvsd/__main__.py", line 45, in <module>
    cli.main()
  File "/home/abolfazl/.vscode/extensions/ms-python.python-2020.1.58038/pythonFiles/lib/python/new_ptvsd/no_wheels/ptvsd/../ptvsd/server/cli.py", line 361, in main
    run()
  File "/home/abolfazl/.vscode/extensions/ms-python.python-2020.1.58038/pythonFiles/lib/python/new_ptvsd/no_wheels/ptvsd/../ptvsd/server/cli.py", line 203, in run_file
    runpy.run_path(options.target, run_name="__main__")
  File "/usr/lib/python3.6/runpy.py", line 263, in run_path
    pkg_name=pkg_name, script_name=fname)
  File "/usr/lib/python3.6/runpy.py", line 96, in _run_module_code
    mod_name, mod_spec, pkg_name, script_name)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/abolfazl/prog/install-dir/mne-python-master/examples/time_frequency/plot_compute_csd.py", line 35, in <module>
    data_path = sample.data_path()
  File "</usr/local/lib/python3.6/dist-packages/mne-0.20.dev0-py3.6.egg/mne/externals/decorator.py:decorator-gen-363>", line 2, in data_path
  File "/usr/local/lib/python3.6/dist-packages/mne-0.20.dev0-py3.6.egg/mne/utils/_logging.py", line 90, in wrapper
    return function(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/mne-0.20.dev0-py3.6.egg/mne/datasets/sample/sample.py", line 21, in data_path
    download=download)
  File "/usr/local/lib/python3.6/dist-packages/mne-0.20.dev0-py3.6.egg/mne/datasets/utils.py", line 395, in _data_path
    remove_archive, full = _download(path, u, an, h)
  File "/usr/local/lib/python3.6/dist-packages/mne-0.20.dev0-py3.6.egg/mne/datasets/utils.py", line 453, in _download
    hash_=hash_, hash_type=hash_type)
  File "</usr/local/lib/python3.6/dist-packages/mne-0.20.dev0-py3.6.egg/mne/externals/decorator.py:decorator-gen-3>", line 2, in _fetch_file
  File "/usr/local/lib/python3.6/dist-packages/mne-0.20.dev0-py3.6.egg/mne/utils/_logging.py", line 90, in wrapper
    return function(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/mne-0.20.dev0-py3.6.egg/mne/utils/fetching.py", line 122, in _fetch_file
    u = request.urlopen(url, timeout=timeout)
  File "/usr/lib/python3.6/urllib/request.py", line 223, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.6/urllib/request.py", line 526, in open
    response = self._open(req, data)
  File "/usr/lib/python3.6/urllib/request.py", line 544, in _open
    '_open', req)
  File "/usr/lib/python3.6/urllib/request.py", line 504, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.6/urllib/request.py", line 1361, in https_open
    context=self._context, check_hostname=self._check_hostname)
  File "/usr/lib/python3.6/urllib/request.py", line 1320, in do_open
    raise URLError(err)
urllib.error.URLError: <urlopen error [Errno 0] Error>
```

agramfort, 2020-02-09:
what's new https://mne.tools/stable/whats_new.html says:

Removed the mne.datasets.megsim dataset because it was taken down by
its host (MRN). Use the mne.simulation functions instead, by Eric
Larson

Ziaeemehr, 2020-02-12:
The problem with `data_path = sample.data_path()` come from connection in my country, apparently there is some restriction access.

closed 2020-02-12 by Ziaeemehr