mne-tools/mne-python #7306: Add legend to `plot_psd`, when using picks

adam2392, 2020-02-11:
#### Describe the problem
When doing `raw.plot_psd(picks=...)`, with multiple axes, and then running `plt.legend()`, the different colors do not show up sufficiently on the legend.

#### Describe your solution
Add labels to each plot on plot_psd, so `ax.legend()` will catch these.

#### Additional context
Add any other context or screenshots about the feature request here.

agramfort, 2020-02-11:
what if you have 300 sensors like MEG?

>

adam2392, 2020-02-11:
Ah I see, sorry, only when you do `average`

```
fig, ax = plt.subplots(1,1)
raw.plot_psd(
               ax=ax, color="blue", picks=stuff,
               average=True, show=False)

raw.plot_psd(
               ax=ax, color="red", picks=other_stuff,
               average=True, show=False)
ax.legend()
```

Ideally, I would presume that the legend shows two sets of labels, colorcoded red and blue?

agramfort, 2020-02-12:
Given how different can be the input of picks parameter I think it’s hard to achieve something consistent. Can you try to pass differently a list of strings to ax.legend function ?