mne-tools/mne-python #7311: Biosemi64 montage and Nz as tip-of-nose

Eklund89, 2020-02-12:
#### Describe the problem
I use biosemi to record 64 electrodes with two additional electrodes: cheek and tip of the nose (**as Nz**).
Previously, I loaded and applied the built-in montage:
`montage = mne.channels.read_montage('biosemi64', unit='mm', transform=True)`
`raw.set_montage(montage, set_dig=True)`
It just informed me that cheek is not part of the biosemi64 montage and thus gets no position.

With the new DigMontage changes, I try using:
`montage = mne.channels.make_standard_montage('biosemi64')`
`raw.set_montage(montage)`

I get this error:
ValueError: DigMontage is a only a subset of info. There are 2 channel positions not present in the DigMontage. The required channels are: ['Nz', 'Cheek']

After removing Cheek, I still get an error for Nz which should be a part of the biosemi64 montage.
Only by removing both cheek and Nz the montage works.

#### Describe your solution
I want to be able to apply the biosemi64 montage to the electrodes that match the default montage and maybe provide additional positions for my other electrodes.

I realize that Nz is a fiducial and should not be used for the tip of the nose. However, its position is not important for its purpose as a reference later. The cheek is used to create a virtual eog channel (Fpz - cheek) and is deleted after preprocessing.

#### Describe possible alternatives
Add a description to the documentation how this can be done. I'm sure its a usual thing.

#### Additional context
Add any other context or screenshots about the feature request here.

larsoner, 2020-02-12:
> I want to be able to apply the biosemi64 montage to the electrodes that match the default montage and maybe provide additional positions for my other electrodes.

I originally thought this was a bug, but I think maybe we shouldn't have had `Nz` as a channel before. Can you see if [make_dig_montage](https://mne.tools/dev/generated/mne.channels.make_dig_montage.html#mne.channels.make_dig_montage) would do what you want? For example you could load the Biosemi montage and use those positions to make a new custom montage that fits your needs (e.g., including an `Nz` EEG channel at whatever location you want).

Eklund89, 2020-02-13:
Thank you for the response @larsoner 
I do not manage to get it to work. Can you help me with this example file?

`import mne`
`file = "sample_raw.fif"`
`raw = mne.io.read_raw_fif(file)`
`montage = mne.channels.make_standard_montage('biosemi64')`
`raw.set_montage(montage)`

[sample_raw.zip](https://github.com/mne-tools/mne-python/files/4197406/sample_raw.zip)

agramfort, 2020-02-13:
@Eklund89 you can do this:

```
import mne
file = "sample_raw.fif"
raw = mne.io.read_raw_fif(file)
montage = mne.channels.make_standard_montage('biosemi64')
raw.set_channel_types({'Nz': 'misc', 'Cheek': 'misc'})
raw.set_montage(montage)
```

basically we need to know the position of all EEG channels to set the montage. biosemi64 does not contain location for Nz or Cheek.

HTH

Eklund89, 2020-03-20:
@agramfort Thank you for the help! Ok so then my next problem. I want to do ICA that includes the Nz channel. I guess the Nz then needs to be changed again to an EEG channel and then I would like to edit its position because it ends up in the ear. How can I do this? The Biosemi montage does have a Nasion fiducial with a position that would work. I just don't know how to set the Nz position. I assume that the ICA needs to have the nose in its true location to do a good job with the spatial filter.

agramfort, 2020-03-20:
@cbrnr or @sappelhoff do you have a code snippet that could be used here? basically code that uses a correct channel file for biosemi. If not I've have a look.

sappelhoff, 2020-03-20:
@Eklund89 this should do the trick. Note, I copy pasted `DATA` from https://github.com/mne-tools/mne-python/blob/master/mne/channels/data/montages/biosemi64.txt

Note how at the end of DATA I add two custom rows, corresponding to your two extra electrodes:

```
MyNas	 115	 90
MyCheek	 -112	 -72
```

In the code snippet I am using a private function, so you probably don't want to make a tutorial with this :-) private functions can always change without notice ...

```Python
from mne.defaults import HEAD_SIZE_DEFAULT
from mne.channels._standard_montage_utils import _read_theta_phi_in_degrees

DATA = """
Fp1	-92	-72
AF7	-92	-54
AF3	-74	-65
F1	-50	-68
F3	-60	-51
F5	-75	-41
F7	-92	-36
FT7	-92	-18
FC5	-72	-21
FC3	-50	-28
FC1	-32	-45
C1	-23	0
C3	-46	0
C5	-69	0
T7	-92	0
TP7	-92	18
CP5	-72	21
CP3	-50	28
CP1	-32	45
P1	-50	68
P3	-60	51
P5	-75	41
P7	-92	36
P9	-115	36
PO7	-92	54
PO3	-74	65
O1	-92	72
Iz	115	-90
Oz	92	-90
POz	69	-90
Pz	46	-90
CPz	23	-90
Fpz	92	90
Fp2	92	72
AF8	92	54
AF4	74	65
AFz	69	90
Fz	46	90
F2	50	68
F4	60	51
F6	75	41
F8	92	36
FT8	92	18
FC6	72	21
FC4	50	28
FC2	32	45
FCz	23	90
Cz	0	0
C2	23	0
C4	46	0
C6	69	0
T8	92	0
TP8	92	-18
CP6	72	-21
CP4	50	-28
CP2	32	-45
P2	50	-68
P4	60	-51
P6	75	-41
P8	92	-36
P10	115	-36
PO8	92	-54
PO4	74	-65
O2	92	-72
Nz	 115	 90
LPA	-115	  0
RPA	 115	  0
MyNas	 115	 90
MyCheek	 -112	 -72
"""

fname = 'chs.tsv'
with open(fname, 'w') as fout:
    fout.write(DATA)

montage = _read_theta_phi_in_degrees(fname=fname, head_size=HEAD_SIZE_DEFAULT,
                                     fid_names=['Nz', 'LPA', 'RPA'],
                                     add_fiducials=False)

```

[image]

cbrnr, 2020-03-20:
I use the standard `biosemi64` montage so I don't have an Nz channel. However, ICA does not need a montage to work (only if you want to visualize results a montage is needed). Would the new standard montage we've been discussing have a suitable location?

cbrnr, 2020-03-23:
Also, I wouldn't include the tip of the nose in ICA because it is unlikely that it contains a lot of brain activity from this location. Same for the cheek location. And even if these two locations did contain significant brain activity, skipping two channels won't degrade the separation significantly.

Eklund89, 2020-03-23:
@cbrnr I use the cheek with Fpz to make an EOG channel to inform the ICA and then I drop the cheek channel so it is not included in the ICA. 

However, if I later want to apply the ICA correction to the nose electrode to remove eyeblink artifacts, don't I need the nose in the ICA?

At the end, I want to use nose as a reference so it needs to be cleaned.

What do you suggest?

cbrnr, 2020-03-23:
I'd use average reference for ICA with 64 EEG plus the nose channels. I don't think you need the cheek channel because you have EEG channels that will pick up ocular activity. Note that if you use the cheek channel it needs to have the same reference as all other channels (so don't compute a bipolar channel).

cbrnr, 2020-04-21:
@Eklund89 I'm closing for now because we've provided some potential answers to your question. Feel free to re-open, or better yet, ask on our mailing list or Gitter channel because this is more a usage question.

closed 2020-04-21 by cbrnr