mne-tools/mne-python #7316: Remove window title decoration from new _TimeViewer

hoechenberger, 2020-02-13:
### Describe the problem
The new `_TimeViewer` by @GuillaumeFavelier  adds its own window title row, which is odd at least on macOS when not running in full screen mode, as the titlebar added by the operating system has the exact same name, so there's now redundant information:

[image]
[image]

#### Describe your solution
The title should be switched off by default, at least on macOS.

GuillaumeFavelier, 2020-02-13:
Thanks for reporting! It might be difficult for me to debug this one at the moment since it happens on macOS but I'll see what I can do. Also, I will link this to #7162

larsoner, 2020-03-04:
@GuillaumeFavelier maybe you can sit with @hoechenberger and get this fixed for 0.20?

hoechenberger, 2020-03-04:
Let's look into this together tomorrow morning, @GuillaumeFavelier

cbrnr, 2020-03-11:
I've never tried this, but shouldn't this issue also affect all other platforms? Window title bars on Windows and Linux also contain names, so I'd assume that this also leads to duplicated titles. It's probably best to just remove the custom titles.

hoechenberger, 2020-03-11:
Good point indeed!

hoechenberger, 2020-03-12:
We couldn't reproduce this on Linux and macOS with different versions of (Py)Qt and PyVista. Seems the issue has magically disappeared since I'd first reported it...
Closing for now!

closed 2020-03-12 by hoechenberger