mne-tools/mne-python #7320: channels.read_custom_montage() silently ignores unit kwarg when loading generic Theta-phi montage file (and possibly others)

hoechenberger, 2020-02-14:
#### Describe the bug
`read_custom_montage()` accepts a `unit` kwarg …
https://github.com/mne-tools/mne-python/blob/d2a5660db534c2d67af514b3d89b08abb67aa696/mne/channels/montage.py#L956-L957

… but doesn't pass the `unit` kwarg down to `_read_theta_phi_in_degrees()`:
https://github.com/mne-tools/mne-python/blob/d2a5660db534c2d67af514b3d89b08abb67aa696/mne/channels/montage.py#L1039-L1044

This is understandable, since `_read_theta_phi_in_degrees()` currently doesn't _accept_ a `unit` kwarg!
https://github.com/mne-tools/mne-python/blob/d2a5660db534c2d67af514b3d89b08abb67aa696/mne/channels/_standard_montage_utils.py#L238-L239

The result is a montage built on the assumption that head size was specified in meters, regardless of the `unit` kwarg the user passes.

#### Steps to reproduce
```python
import pathlib
import mne

base_dir = pathlib.Path('/Users/hoechenberger/Projects/TaDe')
montage_file = base_dir / 'actiCHamp_64Ch_Theta-phi_in_deg_including_fid.txt'

head_sizes = (0.085, 85)
units = ('m', 'mm')

for head_size, unit in zip(head_sizes, units):
    m = mne.channels.read_custom_montage(montage_file,
                                         head_size=head_size,
                                         unit=unit)
    print(f'Specfiied head size: {head_size} {unit}')
    print(f'Resulting LPA loc:\n{m.dig[0]}\n')
```
(I can upload the files if need be, but I think the example is pretty straightforward)

#### Expected results
Both function calls should yield identical results.

#### Actual results
The results differ, as the head size is always assumed to have been specified in meters:
Output:
```
Specfiied head size: 0.085 m
Resulting LPA loc:
<DigPoint |        LPA : (-78.8, -0.0, -31.8) mm   : unknown frame>

Specfiied head size: 85 mm
Resulting LPA loc:
<DigPoint |        LPA : (-78810.6, -0.0, -31841.6) mm : unknown frame>
```

#### Additional information
Since the `unit` kwarg isn't processed correctly, it's also not sanity-checked for invalid values. So one could pass anything, without creating an error message:

```python
mne.channels.read_custom_montage(montage_file, unit='hello world!')
```
Output:
```
<DigMontage | 0 extras (headshape), 0 HPIs, 3 fiducials, 65 channels>
```

A number of other montage-reading functions may be affected too, as they also seem to be assuming the head size is always specified in meters:

- `_read_elc`
- `_read_csd`
- `_read_sfp`
- `_read_elp_besa`

agramfort, 2020-02-14:
PR welcome ! thx for the bug report

hoechenberger, 2020-02-15:
Just for the record, as we figured during the discussion of #7321, this "bug" is based on a misunderstanding on my end: the head size _always_ has to be specified in meters. The `unit` kwarg, otoh, was never really used for anything. #7327 aims to remove it altogether from `read_custom_montage()`.

PR #7327 (2020-02-17): Drop unit kwarg from `read_custom_montage()`
#### Reference issue
Fixes #7320

#### What does this implement/fix?
<strike>Improves the docstring of `read_custom_montage()` to make it clearer that `head_size` has to be specified in meters (and meters only), and that the `unit` kwarg only refers to the montage, not the head.

Also drops the `unit` arg from `_read_eeglab_location()`, as it was unused inside the function.

Currently, the only use of the `unit` kwarg is in the BrainVision reader; but even there, we simply check it the passed value is `auto` or `mm`, and then don't do _anything_ wit [... cut]
closed 2020-02-17 by larsoner