mne-tools/mne-python #7334: watershed_bem error

joshbear, 2020-02-17:
#### Describe the bug
watershed bem file overwrite error / function failure

#### Steps to reproduce
Using both the terminal function, with or without the --overwrite flag:
  mne watershed_bem -s ESES_40
  mne watershed_bem -s ESES_40 --overwrite
Or the function within a Python script: 
  mne.bem.make_watershed_bem()

Returns the following:

#### Expected results
watershed output files for use in subsequent steps

#### Actual results
**From within Python:**
> Saving /Users/joshbear/research/eses_connectivity/fs_anatomy/ESES_40/bem/watershed/ws
> unknown file type for file (/Users/joshbear/research/eses_connectivity/fs_anatomy/ESES_40/bem/watershed/ws)
> done
> mri_watershed utimesec    570.629790
> mri_watershed stimesec    1.718154
> mri_watershed ru_maxrss   657887232
> mri_watershed ru_ixrss    0
> mri_watershed ru_idrss    0
> mri_watershed ru_isrss    0
> mri_watershed ru_minflt   293313
> mri_watershed ru_majflt   715
> mri_watershed ru_nswap    0
> mri_watershed ru_inblock  0
> mri_watershed ru_oublock  0
> mri_watershed ru_msgsnd   0
> mri_watershed ru_msgrcv   0
> mri_watershed ru_nsignals 0
> mri_watershed ru_nvcsw    545
> mri_watershed ru_nivcsw   108991
> mri_watershed done
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "/Users/joshbear/research/eses_connectivity/scripts/preprocess.py", line 430, in preprocess_anat
>     subjects_dir=mri_root_directory)
>   File "</opt/anaconda3/envs/mne/lib/python3.7/site-packages/mne/externals/decorator.py:decorator-gen-49>", line 2, in make_watershed_bem
>   File "/opt/anaconda3/envs/mne/lib/python3.7/site-packages/mne/utils/_logging.py", line 90, in wrapper
>     return function(*args, **kwargs)
>   File "/opt/anaconda3/envs/mne/lib/python3.7/site-packages/mne/bem.py", line 1154, in make_watershed_bem
>     write_surface(s, rr, tris, volume_info=volume_info)
>   File "/opt/anaconda3/envs/mne/lib/python3.7/site-packages/mne/surface.py", line 917, in write_surface
>     fname = _check_fname(fname, overwrite=overwrite)
>   File "/opt/anaconda3/envs/mne/lib/python3.7/site-packages/mne/utils/check.py", line 143, in _check_fname
>     raise IOError('Destination file exists. Please use option '
> OSError: Destination file exists. Please use option "overwrite=True" to force overwriting.

**From Terminal**
Exactly the same as above. 

#### Additional information
>>> mne.sys_info()
Platform:      Darwin-19.3.0-x86_64-i386-64bit
Python:        3.7.5 (default, Oct 25 2019, 10:52:18)  [Clang 4.0.1 (tags/RELEASE_401/final)]
Executable:    /opt/anaconda3/envs/mne/bin/python
CPU:           i386: 16 cores
Memory:        64.0 GB

mne:           0.19.2
numpy:         1.17.4 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.3.1
matplotlib:    3.1.1 {backend=MacOSX}

sklearn:       0.21.3
numba:         0.46.0
nibabel:       3.0.0
cupy:          Not found
pandas:        0.25.3
dipy:          1.0.0
mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.14.0}
pyvista:       0.23.0
vtk:           8.1.2

larsoner, 2020-02-17:
Did you try `master`? This appears to be fixed there already:

https://github.com/mne-tools/mne-python/blob/master/mne/bem.py#L1171-L1172

joshbear, 2020-02-18:
Indeed, it works correctly in master. Thanks and sorry for the bother! 

Josh

closed 2020-02-18 by joshbear