mne-tools/mne-python #7340: White Space in Events_ID from Annotations - EDF file

adam2392, 2020-02-18:
#### Describe the bug

In `events_from_annotations`, I can read in Annotations that were attached to an EDF file read in. However, there are large white-spaces sometimes if the Annotations weren't written properly in the EDF file. 

So instead of: `this happened`, it looks like `this happened                 `. I don't foresee any use case where this would be useful, so I was wondering if I could submit a PR to just `strip()` the whitespace within `annotations._select_annotations_based_on_description`? 

#### Steps to reproduce
Open up edfbrowser, add a bunch of white space to an annotation. Read in via `read_raw_edf` and call `mne.events_from_annotations()`. Then `print(events_id)`.

cbrnr, 2020-02-19:
I agree that normally whitespaces should be ignored, but if we enforce this we're restricting the EDF format unnecessarily. I'd rather leave this to the users to post-process any descriptions. However, if people want that I'd be OK with having an additional parameter for `mne.events_from_annotations` (e.g. `strip_whitespace=False`). What do others think?

larsoner, 2020-02-19:
I think it would be better to provide people ways of cleaning / modifying their annotations rather than `.strip()`ping strings for them. If the annotation in the file is `"this happened "` then it seems like correct behavior that we keep it as `"this happened "` rather than changing it to `"this happened"`.

Can you do something like:
```
for di, desc in annotations.description:
    description[di] = desc.strip()
```
If it's this easy then I think it should be done on the user's side, not the MNE side. And if it's not this easy, we should provide an easy mechanism for replacing annotation descriptions.

agramfort, 2020-02-19:
I would expect users to do the for loop that @larsoner suggested. What if
you want
to enforce lower caps or whatever other string manipulation. Just do a for
loop.

>

cbrnr, 2020-02-19:
I agree with both of you @larsoner and @agramfort, I just wanted to mention that maybe appended whitespace happens very often with EDF. I don't know if that's the case, but only if this is true it would warrant a new parameter. Otherwise, I also think that we should leave any string manipulations to the user.

@larsoner the descriptions are in a NumPy array, so a simple list comprehension might work in many cases:

```
raw.annotations.description = [d.strip() for d in raw.annotations.description]
```

However, this will create a list, which shouldn't be a problem. This won't work if you want to add characters though.

larsoner, 2020-03-04:
Let's close for now then