mne-tools/mne-python #7345: Q: _TimeViewer interaction

kingjr, 2020-02-19:
Hi @GuillaumeFavelier, 

I'm trying to test your beautiful `TimeViewer`. It works nicely: I can move the camera and play the animation by pressing the space bar, but I don't manage to click / drag the cursors. I think I'm missing something silly, any idea?

Thanks!

kingjr, 2020-02-20:
I should add that I'm on macOs, and using pyvista 0.23.1

GuillaumeFavelier, 2020-02-20:
>  but I don't manage to click / drag the cursors. 

You mean the sliders? This is odd. Can you share the result of :

`python -c "import mne; mne.sys_info()"` 

and

`python -c "import pyvista; print(pyvista.Report())"`?

My first suggestion would be to try with the `master` branch of `pyvista`. Let me know how it goes.

kingjr, 2020-02-20:
<string>:1: RuntimeWarning: macOS users should use PyQt5 >= 5.10 for GUIs,
got 5.9.2. Please upgrade e.g. with:

    pip install "PyQt5>=5.10"

Platform:      Darwin-19.2.0-x86_64-i386-64bit

Python:        3.7.4 (default, Aug 13 2019, 15:17:50)  [Clang 4.0.1
(tags/RELEASE_401/final)]

Executable:    /Users/jeanremi/opt/anaconda3/bin/python

CPU:           i386: 12 cores

Memory:        32.0 GB

mne:           0.20.dev0

numpy:         1.17.2 {blas=mkl_rt, lapack=mkl_rt}

scipy:         1.3.1

matplotlib:    3.1.1 {backend=MacOSX}

sklearn:       0.21.3

numba:         0.45.1

nibabel:       2.5.1

cupy:          Not found

pandas:        0.25.1

dipy:          Not found

mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.9.2}

pyvista:       0.23.1

vtk:           8.1.2

--------------------------------------------------------------------------------

  Date: Thu Feb 20 11:19:18 2020 CET

            Darwin : OS

                12 : CPU(s)

            x86_64 : Machine

             64bit : Architecture

           32.0 GB : RAM

            Python : Environment

        GPU Vendor : ATI Technologies Inc.

      GPU Renderer : AMD Radeon Pro 555X OpenGL Engine

       GPU Version : 4.1 ATI-3.4.19

  Python 3.7.4 (default, Aug 13 2019, 15:17:50)  [Clang 4.0.1

  (tags/RELEASE_401/final)]

            0.23.1 : pyvista

             8.1.2 : vtk

            1.17.2 : numpy

             2.6.0 : imageio

             1.4.3 : appdirs

             0.5.0 : scooby

             3.1.1 : matplotlib

             5.9.2 : PyQt5

             7.8.0 : IPython

             1.0.0 : colorcet

  Intel(R) Math Kernel Library Version 2019.0.4 Product Build 20190411 for

  Intel(R) 64 architecture applications

--------------------------------------------------------------------------------

larsoner, 2020-02-20:
Can you try updating PyQt as suggested by the warning? There are interactivity problems with older PyQt on macOS.

kingjr, 2020-02-20:
Yes, updateing PyQt5 fixed it thanks

closed 2020-02-20 by kingjr