mne-tools/mne-python #7347: mne.Info - offer more complete options for subject handedness

dominikwelke, 2020-02-21:
dear all, 
first of all thanks for your amazing work!

I find in the [api documentation](https://mne.tools/dev/generated/mne.Info.html#mne.Info) that `mne.Info` objects currently only allow two options for a subjects handedness:

`hand : int`
`Handedness (1=right, 2=left).`

is there a specific reason for this? otherwise I would propose adding additional / more complete options:

`hand : int`
`Handedness (0=n/a, 1=right, 2=left, 3=ambidextrous).`

best, Dominik

larsoner, 2020-02-22:
We follow the official FIFF definitions, so we would need to add it there:

https://github.com/mne-tools/fiff-constants/blob/064604db8aeba310c76c93e7c76cc983257105b4/DictionaryTypes.txt#L453-L457

Okay if I transfer the issue there?

hoechenberger, 2020-02-23:
+1 on moving it over there.

btw I believe a similar adjustment could and should be made to the "sex" variable. Will open an issue over at fiff-constants too.

dominikwelke, 2020-02-23:
@larsoner yes, this would be great. thanks for forwarding this to the right address!

@hoechenberger +1 on adding "sex" options, I thought of this as well