mne-tools/mne-python #7350: Setting custom_ref_applied for ECoG data

libertyh, 2020-02-23:
#### Setting average reference for ECoG data
When using the `raw.set_eeg_reference(ref_channels='average')` to perform a common average reference on ECoG data, the `custom_ref_applied` bool flag in raw.info remains set to `False`. I looked at `io/reference.py` and in the function definition for `set_eeg_reference` it appears that when `_apply_reference` is called, it looks only for channels of type 'eeg' is 'custom_ref_applied' set to `True`. 

Also, in the comments for `_apply_reference`, it appears to say that `info['custom_ref_applied']=True` refers to a "non-CAR" reference, but the test is whether the reference includes EEG electrodes (which it would for an average reference, right?)

My general question then, is, for consistency, should `custom_ref_applied=True` after applying an average reference to ECoG data, or am I misunderstanding the use of this flag?  It seems like a nice marker for whether this operation has happened or not. Happy to help fix if this makes sense.  I see this is similar to this issue: https://github.com/mne-tools/mne-python/issues/6454, but not exactly.

agramfort, 2020-02-24:
what type of reference would you apply on ECoG data? CAR? The usecase with
ECoG or sEEG has been
really explored that why it's called set_eeg_reference and not
set_reference(ch_type='eeg')

I guess we 'just' need the use case to adjust MNE to behave properly.

libertyh, 2020-02-24:
For ECoG/sEEG, a CAR is common, but not always done. Sometimes a single reference channel is used (a subgaleal reference, for example, or even a white matter reference). Bipolar reference is also often used clinically. Basically it would make sense for `set_reference` or `set_eeg_reference` to work exactly as it does, but have the ability to set the channel type, as you suggested above.

agramfort, 2020-02-24:
ok you suggest we need a set_reference(ch_type) and make set_eeg_reference
call set_reference(ch_type='eeg') before deprecating set_eeg_reference()

any objection anyone? Can you give it a shot?

>

larsoner, 2020-02-24:
Rather than force a deprecation cycle, I would just add a `ch_type` argument to `set_eeg_reference`. It leaves it as a misnomer but since `set_eeg_reference` is mandatory for inverse modeling, it's going to be all over the place for users so the pain of deprecating this function is quite high.

hoechenberger, 2020-02-24:
> Rather than force a deprecation cycle, I would just add a `ch_type` argument to `set_eeg_reference`. It leaves it as a misnomer but since `set_eeg_reference` is mandatory for inverse modeling, it's going to be all over the place for users so the pain of deprecating this function is quite high.

In that case, I'd prefer @agramfort's proposal sans the deprecation… i.e., add `set_reference()` and simply make `set_eeg_reference()` call `set_reference(ch_type='eeg')`

drammock, 2020-02-24:
> Rather than force a deprecation cycle, I would just add a `ch_type` argument to `set_eeg_reference`. It leaves it as a misnomer [...]

Could we compromise by setting a longer deprecation cycle? (2 releases? Or even 4?)

libertyh, 2020-02-24:
FWIW, I don't feel strongly about it being called `set_reference` instead -- intracranial EEG is a subset of EEG, I guess, so semantically it doesn't bother me if that's also the issue.

larsoner, 2020-02-24:
> Could we compromise by setting a longer deprecation cycle? (2 releases? Or even 4?)

To me it's not about the duration, it's about the cost (amount of code changes people will have to make at some point or another) relative to the gain we (maintenance) and users (usability) get from the change. Here what I see is no maintenance benefit, a tiny usability/discoverability benefit, and a high cost of changing (i.e., any user scripts that involve EEG inverse modeling, or even most that process EEG in general, since these often set references with or without projections).

agramfort, 2020-02-24:
I anticipate people will want to do it for sEEG too as some point...

>

agramfort, 2020-02-24:
I am fine keeping set_eeg_reference to avoid breaking too much code

>

libertyh, 2020-02-24:
Ok I can take a stab at this later!

larsoner, 2020-03-04:
@libertyh just a heads up that we're looking to release 0.20 in the next couple of weeks so if you want this fix to make it into that release we'll have to iterate somewhat quickly. No rush from our end, though

PR #7393 (2020-09-09): BUG, ENH: Added ch_type option to _apply_reference so custom_ref_applied bool b…
#### Reference issue
Example: Fixes #7350 

#### What does this implement/fix?
This changes `_apply_reference` in `mne/io/reference.py` so that `custom_ref_applied` is set to True for ECoG and SEEG electrodes (not only EEG electrodes as before). 

#### Additional information
In previous discussions of this change, it was suggested to add `ch_type` to `set_eeg_reference`, but this keyword argument already exists and so did not need to be added. It turns out that `ch_type` was not being used in `_apply_reference` wh [... cut]
closed 2020-09-09 by larsoner