mne-tools/mne-python #7359: ENH: what to do with muscle artifacts

AdoNunes, 2020-02-27:
If I am not mistaken, currently, to remove muscle artifacts it is only possible by manually annotating bad segments or write custom code to detect them.

Muscle artifacts are an issue if data is not low band passed, i.e. if studying gamma activity. 

If you think that a function for estimating segments of data with muscle artifacts would be a good addition, let me know and I will make a PR.

agramfort, 2020-02-27:
Before starting a PR can you draft the API by providing a few lines of usage how you envision it? Thx

AdoNunes, 2020-02-27:
I was thinking to be similar to the head motion detection. Basically, the input is the raw data and the output the annotation of segments with muscle artifacts.

The function would be very similar to how Fieldtrip does it. The data is band passed filtered between 110-140Hz, z-scored, low-passed to reduce spikes, z-score averaged across channels, and whatever is above the z-score threshold get annotated.

drammock, 2020-02-28:
@AdoNunes can you mock up the API *exactly as you envision it*?  Something like

```python
name_of_output = mne.preprocessing.name_of_new_function(arg1, arg2)
```

and if there are variable outputs, give additional examples, like 

```python
(output1, output2) = mne.preprocessing.new_function(arg1, arg2, return_extra_output=True)
```

AdoNunes, 2020-02-28:
Actually I did a PR #7096 with this function that at the end got shortened to `annotate_movement`.
The API looks like this:

``` threshold_muscle = 1.5  # z-score
annotation_muscle, scores_muscle = mne.preprocessing.annotate_muscle(raw, thr=threshold_muscle, t_min=0,
                                                 notch=None)
```

Returns annotations and a 1xn_samples vector with the averaged z-scores at each sample.

agramfort, 2020-02-28:
> annotation_muscle, scores_muscle = mne.preprocessing.annotate_muscle(raw, thr=threshold_muscle, t_min=0,
>                                                  notch=None)
>
> Returns annotations and a 1xn_samples vector with the averaged z-scores at each sample.

I would call parameter threshold not thr. Why do you need notch? what
is t_min? if onset of raw why not pass
a cropped raw? We should aim for the simplest API that does the job.

larsoner, 2020-02-28:
For example if notch filtering is optional (e.g., to remove line frequencies), users should do it before calling the function and we can tell them to in the notes section.

AdoNunes, 2020-02-28:
OK. 
Then the API should look like this:
```
annotation_muscle, scores_muscle = mne.preprocessing.annotate_muscle(raw, threshold =threshold_muscle)
```
Can you suggest me a dataset reach in muscle artifacts for the example and tests?

agramfort, 2020-02-28:
maybe some of EEG BCI dataset we offer in some examples?

>

AdoNunes, 2020-03-02:
It might be necessary to add the pick types (or channels) and also set a minimum segment length of good segments, i.e. mark as bad a segment between two annotations shorted than, for example, .2 seconds. 
```
annotation_muscle, scores_muscle = mne.preprocessing.annotate_muscle(raw, threshold,
picks, min_length_good)
```
What do you think?

larsoner, 2020-03-02:
Sounds reasonable to me

drammock, 2020-03-02:
Is `annotate_muscle` is the right name for the function (compared to, say, `find_muscle_artifacts` or `annotate_muscle_artifacts` or something else)?  I don't have strong instincts about what is best here, just want to ask the question now, before it becomes an API change question.

drammock, 2020-03-02:
also wondering whether this function catches all kinds of muscle artifacts (swallowing, chewing, facial/scalp muscle, tongue/jaw/speech, etc).  If not, that might suggest a more narrow naming choice like e.g. `annotate_swallowing_artifact`.

AdoNunes, 2020-03-02:
@drammock I named to be consistent with `annotate_muscle`. Yes, it caches all types of muscles around the head as long as they are strong enough.

PR #7407 (2020-04-07): ENH: muscle artifact detection
closes #7359 

I have added a new function `detect_muscle` in artifact_detection.py. I also included the example and test.
closed 2020-04-07 by larsoner