mne-tools/mne-python #7385: MAINT: Release 0.20

larsoner, 2020-03-04:
We are due to release 0.20, and plan to shoot for releasing in the next couple of weeks. As some have probably noticed, I did a bit of issue culling and milestone tweaking. This is the current list of issus/PRs marked for 0.20:

https://github.com/mne-tools/mne-python/milestones/0.20

@mne-tools/contributors @mne-tools/admins feel free to go through the PRs/issues and mark ones you think should make it into 0.20 (or if you don't have permissions, ask that I mark them), keeping in mind that we only have a couple of weeks to get things merged.

yh-luo, 2020-03-17:
Hi, I previously found and fixed a bug. Do I need to mark that  PR if it was merged to master and closed?

agramfort, 2020-03-17:
no we need only to list what remains to be merged and closed

>

yh-luo, 2020-03-17:
got it. thanks!

drammock, 2020-03-19:
We are preparing to release MNE-Python 0.20.  There are several contributors who do not have a (public) name associated with their account, and hence will be listed only by their GitHub username. If you would prefer your MNE-Python contribution to be credited under your actual name, please respond here:

@carlos
@dominikwelke
@fmamashli
@foucault
@klankinen
@kostasde
@mdaneshzand
@mohdsherif
@olafhauk

And thanks for contributing to MNE-Python!

agramfort, 2020-03-19:
for some of them it's pb in the mailmap. See diff:

diff --git a/.mailmap b/.mailmap
index fe5bd308b..1f9e6d1e5 100644
--- a/.mailmap
+++ b/.mailmap
@@ -53,6 +53,7 @@ Marmaduke Woodman <mmwoodman@gmail.com> maedoc <
maedoc@mm.st>
 Brad Buran <bburan@galenea.com> Brad Buran <bburan@alum.mit.edu>
 Cathy Nangini <cnangini@gmail.com> CN <cnangini@gmail.com>
 Olaf Hauk <olaf.hauk@mrc-cbu.cam.ac.uk> Olaf Hauk <olaf@mac0086.local>
+Olaf Hauk <olaf.hauk@mrc-cbu.cam.ac.uk> olafhauk <
olaf.hauk@mrc-cbu.cam.ac.uk>

to find out:

git rev-list -n 1 v0.19.0 | xargs -I commit git log commit.. | git shortlog
-se | sed "s/^[      ]*[0-9]*[        ]*/* /"

>

dominikwelke, 2020-03-19:
thanks for asking. I updated my profile..

mohdsherif, 2020-03-20:
Updated mine as well (mohdsherif)
Thanks.

fmamashli, 2020-03-20:
Thanks to note this. I updated!

klankinen, 2020-03-20:
I updated mine as well, thank you!

larsoner, 2020-03-27:
Email should be sent out imminently so I'll close

closed 2020-03-27 by larsoner