mne-tools/mne-python #7404: BUG topomap grads

agramfort, 2020-03-07:
```
import numpy as np
import mne
from mne.datasets import sample
data_path = sample.data_path()
raw = mne.io.read_raw_fif(data_path + '/MEG/sample/sample_audvis_raw.fif')
info = raw.load_data().pick('grad').info
data = np.random.randn(204)
mne.viz.plot_topomap(data, info)
```

produces 
[image]

mmagnuski, 2020-03-08:
That's an artifact of `extrapolate='box'` - it should look better with `extrapolate='head'` or `extrapolate='local'`.

agramfort, 2020-03-08:
head:
[image]

local:
[image]

so not much success.

agramfort, 2020-03-08:
@larsoner any suggestion?

mmagnuski, 2020-03-08:
Ah, I misunderstood your initial issue. I thought the circle shows what is wrong, but it is the head outline. It is weird indeed, I'll take a look at this.

mmagnuski, 2020-03-08:
It seems that `_pair_grad_sensors` does not take `sphere` as an argument and no sphere-placement is done in general with grad ch type. I'll try fixing that later today.

agramfort, 2020-03-08:
indeed the bug only appears with grad not mag.

thanks for taking a look

>

mmagnuski, 2020-03-08:
@agramfort @larsoner 
I'm not sure what should be considered the source of this bug (apart from ignoring `sphere` argument):
* with 'grad', channel positions in info are ignored, only the layout is used
* layout for `'Vectorview-grad'` is not zero-centered, see below:

[image]

agramfort, 2020-03-08:
one should use the same thing as for mag. One should not use layout also in
this case.

>

PR #7411 (2020-03-09): FIX: topomap with grad ignored sphere and used layout (#7404)
@agramfort - see if this works for you (it does for me).
fixes #7404 #7406

###  TODO:
- [x] add test
- [x] no need for whatsnew update, right?
commit 9f05d240 (2020-03-09): FIX: topomap with grad ignored sphere and used layout (#7404) (#7411)

* FIX: fixes #7404

* TST: test for topomap channel placement with ch_type='grad'

* STY: make agramfort happy

* Update test_topomap.py

* STY: make everyone happy, hopefully
closed 2020-03-09 by agramfort