mne-tools/mne-python #7429: Annotations repr too long

cbrnr, 2020-03-12:
I don't like that the `repr` for an `Annotations` object exceeds 79 characters. For example, the following repr has 94 characters:

```
<Annotations  |  30 segments : T0 (15), T2 (7), T1 (8), orig_time : 2009-08-12 16:15:00+00:00>
```

It would be nice if the maximum character length of this (and any other) repr was 79, otherwise these get hard to read because of line breaks.

In the specific case of `Annotations`, I would remove the `orig_time` information because I don't think this is important enough to show in the repr.

agramfort, 2020-03-12:
maybe just give time as "Time: 2009-08-12" to omit too precise info?

>

cbrnr, 2020-03-12:
> maybe just give time as "Time: 2009-08-12" to omit too precise info?

I don't think this is that important. I'd rather use the available characters to show more annotations.

agramfort, 2020-03-12:
we have some repr that are multilines I think

cbrnr, 2020-03-12:
Yes, for example `info`. We also have some reprs that definitely exceed 79 characters (not just `Annotations`).

drammock, 2020-03-12:
If there are a lot of different annotation labels, you can't guarantee they'll all fit even if you remove time completely. Will you truncate then? Or change to a summary message like "30 segments, 21 labels"?

(BTW I agree that the time is not that important to include in the repr here)

agramfort, 2020-03-12:
do what you think is best. I don't have strong feelings here.

>

PR #7441 (2020-03-13): Shorten repr of Annotations if necessary
Fixes #7429.

Before (94 characters):
```
<Annotations  |  14 segments : a (1), b (1), c (1)..., orig_time : 2013-11-04 14:35:26.150000>
```

After (76 characters):
```
<Annotations | 14 segments: a (1), b (1), c (1), d (1), e (1), f (1), g ...>
```
closed 2020-03-13 by agramfort