mne-tools/mne-python #7430: Channels incorrectly scaled in `ica.plot_components()`

cbrnr, 2020-03-12:
Channel locations are now incorrectly scaled when plotting ICA components with `ica.plot_components()`. Here's what the standard 10-20 layout looks like:

[image]

This used to work previously, see e.g. here: 

[image]

Interestingly, plotting the montage works:

```python
montage = mne.channels.make_standard_montage("standard_1020")
montage.plot()
```

[image]

larsoner, 2020-03-12:
Do you only have channels C5-C6 for example in the middle row? If so it looks like the subselection might be correct (?). But if you have all channels from the 10-20 montage and some are just missing, then that's clearly wrong...

cbrnr, 2020-03-13:
Hm, I use these channels:

```
["Fz", "FC3", "FC1", "FCz", "FC2", "FC4", "C5", "C3", "C1", "Cz",
 "C2", "C4", "C6", "CP3", "CP1", "CPz", "CP2", "CP4", "P1", "Pz",
 "P2", "POz", "EOG1", "EOG2", "EOG3"]
```

Fz is the most frontal channel, so it looks like the locations are indeed correct, or at least they match the locations in the montage. However, aren't these locations a bit off? I'd expect Fz to be much more frontal on the topomap, or POz much more occipital. Somehow, the previous figure looks more correct to me (which means that in the new montage plot, more channels should be located outside the head circle - I'm assuming that the head border is at the level of the inion at the most).

larsoner, 2020-03-13:
Closing as dup of #7141, see and/or continue discussion there

closed 2020-03-13 by larsoner