mne-tools/mne-python #7436: Reading single epochs from a fif file

hubertjb, 2020-03-12:
I noticed that reading single epochs is significantly slower when the raw data is saved as a fif file vs. an edf file. Also, the reading time from fif files linearly increases with how far in the recording the epochs are, which is not the case with edf files.

I'm interested in getting this reading time as low as possible since I have to repeatedly load single epochs. I was wondering whether this difference between fif and edf is expected (maybe part of it is due to the lower precision of edf files).

#### Steps to reproduce
I created a gist that creates .fif, .edf and .hdf5 files out of the same fake raw data, and then times the loading of each individual epoch: https://gist.github.com/hubertjb/1b434688dba296d25eba8fab2866a50f

I included direct reading from an hdf5 file as a baseline. 

#### Expected results
I was expecting fif and edf to yield similar results. Specifically, I was expecting the loading time to be more or less constant across the recording for fif. 

#### Actual results
As shown below, reading from fif is slower than from edf, and reading times increase as the epochs get further from the beginning of the recording.
[image]

Mean loading times are
{'edf': 0.0028056514029409373,
 'fif': 0.008060473086787204,
 'hdf5': 0.0009250746053807876}

#### Additional information
Platform:      Linux-4.15.0-70-generic-x86_64-with-debian-buster-sid
Python:        3.7.6 (default, Jan  8 2020, 19:59:22)  [GCC 7.3.0]
Executable:    /storage/store/work/hjacobba/miniconda3/envs/braindecode/bin/python
CPU:           x86_64: 72 cores
Memory:        376.5 GB

mne:           0.20.dev0
numpy:         1.18.1 {blas=openblas, lapack=openblas}
scipy:         1.4.1
matplotlib:    3.1.3 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.22.1
numba:         0.48.0
nibabel:       Not found
cupy:          Not found
pandas:        1.0.1
dipy:          Not found
mayavi:        Not found
pyvista:       Not found
vtk:           Not found

@agramfort @larsoner

larsoner, 2020-03-13:
> I was expecting fif and edf to yield similar results. Specifically, I was expecting the loading time to be more or less constant across the recording for fif.

In FIF the raw data are stored in "tags" of (usually) 1 sec duration so in principle I don't see why it should be slower. I agree it's worth looking into. @hubertjb do you want to look into it? If not I can give it a shot

larsoner, 2020-03-13:
... actually from looking at the FIF code I already suspect a culprit. I'll open a WIP PR.

PR #7448 (2020-03-14): MRG, ENH: Raw data reading refactor
In `mne/io/fiff/raw.py` we iterate over tags to figure out which ones to read. This will slow down as you access data later and later in the file, so a better strategy is to construct arrays of starts/stops and then figure out which tags to read based on that (a simple `searchsorted` or even `np.where` will yield the answers). 

First step is then to refactor the reading function to "reformat raw_extras to be a dict of list/ndarray rather than list of dict (faster access)", which is done here. Hopefully all is green and we can move on to the speedup part.

A [... cut]
closed 2020-03-14 by larsoner